prusa3d/Prusa-Firmware #2242: G92 X Y Z offset bug

MandEdesign, 2019-09-30:
Hello all.  I have been trying to print on top of a jig that I have permanently fixed to the Spring Steel Sheet.  The goal is to use G92 to offset the coordinate system globally referencing the top lower left hand corner of the jig.   

I noticed this weird issue when issuing a G92 X0 Y0 Z0,  it actually seems to set the X and Y a few MM +X, -Y and very slightly -Z.   The axis are not moving when I issue the command, also G92.1 is not supported so I have to home G28 W in order to change back to actual machine home coordinates. 

I know that G92 E0 is issued all the time in slicer code so I figured this would also work for X Y and Z.   Any help on this potential bug here would be greatly appreciated.

wavexx, 2020-01-27:
How are you homing the machine? Are you using G80 (mesh leveling)?
G28 X Y does also alter the Z offset if MBL is active, and I'm not sure this is intentional: this is  different compared to what the official marlin docs say.

A picture of the fixture would be helpful

Prusa-Support, 2022-06-04:
This issue will be closed as stale, and because of the lack of response.

Michele Moramarco
Prusa Research

closed 2022-06-04 by Prusa-Support