prusa3d/Prusa-Firmware #2248: Speeding up the scrolling filename on MK3S LCD will eventually change >>Speed setting

jlastofka, 2019-10-02:
I've discovered I can spin the control knob (without clicking it) and speed up the scrolling of the filename at the bottom for quicker reading. However, if I keep this up for a bit it eventually starts changing the >>100% speed to different values. This is without any clicks or deliberate menu selections. It certainly seems like a bug. This is on a brand new MK3S kit I just purchased and assembled 28 Sep 2019. It has the latest firmware 3.8.0

leptun, 2019-10-02:
Are you using firmware compiled by yourself or precompiled firmware from the website/github?

jlastofka, 2019-10-02:
Mine came installed on the new Prusa MK3S printer I just built. I just learned on the Prusa forum what's going on, more or less. The knob is supposed to control the speed when you turn it from the main info screen. I didn't see that happening because I was only turning the knob several clicks at a time. If I crank it about 3/4 turn it wakes up and does the speed changes. Maybe mine's acting different because of the mile long filenames I'm using on the SD card. I had read the entire book from Prusa and it showed the speed command under the Tune menu so when they talked in the book about turning the knob to adjust the speed I thought that's where they meant. I see now that it can be done from the main screen without the menu. Sorry I may have confused people with my observations, but I got there honestly:-)

closed 2019-10-02 by jlastofka

leptun, 2019-10-02:
Ok. I thought you meant the “Print From SD” menu. :)