prusa3d/Prusa-Firmware #2249: Power panic issue

Makrousek, 2019-10-02:
When power panic occurs during printer pause (for example if there is an issue with MMU, the printer pauses and asks for user intervention), then after power is restored the following occurs:

1) hotend does not heat (can be solved manually by setting "load to nozzle" - when nozzle starts heating, it continues even after pressing cancel on the "load to nozzle" screen)
2) Printer considers Z axis to be on the last layer printed, whilest it is in fact higher (because when pausing for user intervention, z is lifted slightly and x is moved to 0). This causes print to fail, because even though X and Y are reset correctly, Z is too high when printing resumes.

MonoSann, 2019-10-03:
I have a simmilar issue with the I3 MKS without MMU.
When pausing manually during print, and powerloss occures, 
resuming is not possible. Printer is in the standard menue mode.

michalxfanta, 2019-10-09:
This is a known issue and we will plan to fix it.

closed 2023-07-18 by DRracer