prusa3d/Prusa-Firmware #2258: Not able to change language

Zarkex, 2019-10-04:
https://youtu.be/yaLel_mRAeA
As you can see if i try to change language it goes ahead and "changes it" but after that it tries to ask to select a language but if i select the same language again it starts all over again.
If i choose English though it leaves the Menu.

3d-gussner, 2019-10-06:
Hi @Zarkex 
did you use the pre-compiled hex files from Prusa and uploaded it via PrusaSlicer?
If you did, try a [factory rest](https://help.prusa3d.com/article/SYvbQ66IXF-factory-reset).
If that doesn't help I suspect that your external flash has some issues, BTW which Einsy board version do you have?

Or did you compile it yourself?
If you compiled it yourself:
- Did you use Arduino IDE ?
- Did you use `PF-build.sh` ?

I am asking because if you want to compile your firmware yourself the `README.md` points out for Arduino IDE only you have to change [`LANG_MODE` to `0`](https://github.com/prusa3d/Prusa-Firmware#2-source-code-compilation), which is English only.

To get multiple languages you need to run `PF-build.sh` or follow some procedures to get it done manually.

Zarkex, 2019-10-06:
Did all of those things. Used one of the pre-compiled hex files. Flashed via avrdude so that should be correct. Have tried a Factory Reset already.

leptun, 2019-10-06:
Firmware must be flashed with PrusaSlicer if it is multi-language. It requires a modded avrdude for it to work which is the one integrated into PrusaSlicer. Otherwise the firmware uploads correctly, but the language files don't and that's why it fails changing the language.

3d-gussner, 2019-10-06:
@Zarkex as @leptun and I pointed out you need to flash the multi language firmware using PrusaSlicer at this moment.
The default avrdude doesn't write the multi language data to the external flash of the Einsy.

3d-gussner, 2019-10-08:
@Zarkex Hi, did you try to upload the firmware with Prusa Slicer? Any news?

Zarkex, 2019-10-08:
Jup. It worked. Sorry for the "Issue". The more you know :)

leptun, 2019-10-08:
If that’s the case, the issue can be closed.

closed 2019-10-08 by michalxfanta