prusa3d/Prusa-Firmware #2259: Printer mis-interprets g-code and crashes

sfeine, 2019-10-05:
When attempting to print the attached g-code my MK3 (not 's') running 3.8.0 will crash it's y-axis when it gets to about line 1772 (not necessarily this line exactly). This g-code was prepared using the latest version of Prusa Slicer. 

I have tried re-slicing this file 3 times with slightly different settings each time (and different part locations on the bed). All 3 times it has failed in a similar manner.

In all 3 cases the printer was running in silent mode.
I am using 7x7 bed leveling on this printer.
When this part fails the head overshoots the part on a diagonal until an axis crashes.
Then the printer attempts to continue printing with a shifted layer.

I have examined this g-code using simplify 3d and nothing appears to be wrong with it.
I have not yet tried re-flashing the printer's firmware.

My best guess is that the firmware is mis-interpreting the g-code or getting stuck on a math error during this head movement but I am likely wrong. 

I noticed similar behavior on another part that I attempted to print twice earlier in the week. 
I have however printed 3 perfectly good parts in between the parts that keep failing.

[image]

[image]

Any help is appreciated.

[hull_front_l_MIR (1)_0.2mm_PLA_MK3_13h37m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/3692752/hull_front_l_MIR.1._0.2mm_PLA_MK3_13h37m.gcode.zip)

sfeine, 2019-10-05:
Update:

After reverting to 3.7.2 printing the exact same g-code as was posted (and failed in the first photo) appears to work with no issues.

leptun, 2019-10-06:
Thanks for reporting.
Please try the following: disable filament sensor and try to reproduce the issue again.

amherms, 2019-10-06:
I'm am having a similar issue with 3.8.0 on an original i3 Mk3, usually on parts with larger surface area and longer gcode tool movements. I have found that turning down the print speed to 70% usually allows the prints to succeed. When it fails, the printer stops dead in the middle of a tool movement pauses for a second and then continues, running right past the boundary of the print until it hits the physical limit of the printer and the steppers grind. After that it has no idea where it is and keeps printing with both X & Y offset.
[image]

mherms, 2019-10-07:
@leptun suggestion to turn off the filament sensor seems to successfully workaround the bug. Hopefully the next firmware update will allow use of the filament sensor again to avoid unrecoverable prints when a spool runs out.

leptun, 2019-10-07:
That confirms my theory. The print stops because the filament sensor is not sure if there is filament so it stops, moves the filament back and forth and determines that there is filament. At that point it tries to continue, but for some reason it just goes crazy and overshoots.

Maybe running in stealth mode just uncovers a flaw in the filament check resume for the MK3 because it is triggered more often because of the reduced acceleration (just a theory)

DRracer, 2019-10-11:
@leptun your theory was correct, we managed to simulate a similar behavior on an MK3. It is caused by the filament sensor trying to recover from a possible filament runout. 
@sfeine thanks for reporting, we'll fix that

wavexx, 2019-10-20:
Relevant PR #2274 for the intrepid

vintagepc, 2019-10-27:
This seems very similar to #1522

wavexx, 2019-10-27:
Indeed, that's a good catch. And similarly, the pause seem to be triggered by the filament sensor.
This should be fixed properly with the above PR.

Thinking out loud, I now think that the long extrusion in itself (along the diagonal) might be overflowing the some of the internal counters, causing a false trigger. This is something that also brings both issues together, but I cannot believe the optical sensor is failing over a long extrusion

vintagepc, 2019-10-28:
When I've seen it happen and it doesn't run off the edge of the plate, it results in a severe layer shift that (eyeballed) looks about the length of the extrusion should have been, almost as if the sensor trigger caused it to re-do the last extrusion command from scratch (e.g. the filament interrupt is not handled gracefully w/r/t the buffer). Normal filament change (tune menu) would seem to wait until any planned moves are done, but of course this luxury doesn't exist on runout. 

It's definitely an issue with the long extrusions as I never saw this on small prints, it was always in the middle of a very long straight shot at high speed. Is it possible that the planner/motion control is pre-empting the filament sensor timing/handling and causing it to think it ran out from lack of motion when in fact it was just a lack of sensor handling?

Should be pretty easy to repro, just print a large flat square the full size of the build plate. It'll happen during the high speed infill after layer 1.

mherms, 2019-11-02:
Does this issue only affect the MK3 firmware? I have the MK3S upgrade kit but have not installed it yet. It would be worth stopping my backlog of prints to upgrade if I could turn my filament sensor back on.

vintagepc, 2019-11-02:
> Does this issue only affect the MK3 firmware? I have the MK3S upgrade kit but have not installed it yet. It would be worth stopping my backlog of prints to upgrade if I could turn my filament sensor back on.

As far as I can tell it is a result of the Mk3 optical sensor design triggering a false runout. The 3S IR design (when properly adjusted) is much less prone to false runouts and therefore far less susceptible to this issue, if at all. I've seen the odd report of it with a 3S but I think those turned out to be issues with the sensor occasionally reading 0 due to undersized filament or something.

(Note I'm not an official firmware dev, just offering insight based on forum threads, my own experience with this issue, various extruder upgrades to circumvent it, and finally a 3S upgrade)

wavexx, 2019-11-02:
In this case, the issue affects both. In the MK3 a false trigger is just more likely, but an MK3S runout trigger can also result in the same crashing behavior.

I will be probably on the unpopular side, but I'm quite impressed by the MK3 sensor. Contrarily to what is being written, I have excellent tracking on all materials, including completely clear PETG. Some additional sensor fixes coming with LA1.5 will also furthermore reduce the number of false triggers and I'm trying to investigate the behavior with long fills as well.

vintagepc, 2019-11-02:
I had a good experience with it as well after I upgraded to a skelestruder with a dual-bearing-and direct mod (i.e. bearing and filament in field of view) to cut down on false triggers. I probably would have kept it in that setup had I not installed an MMU2S. (I tried running as a Mk3 but gave up to reduce the number of variables in play while getting the setup working and tuned in)

leptun, 2019-11-02:
@wavexx I agree with you. I've only had the sensor malfunction when I was printing black PETG parts, but it turns out I also had the sensor covered in dust. I think that for normal MK3 usage the sensor is enough as it is now + some software improvements. On the other hand, using this sensor with the MMU2 is probably a nightmare. That's why I would have hoped Prusa chose to include both sensors for the MMU2S, having an optical sensor that detects jams, while having the IR sensor detecting filament at gears. The mk3 could have both of them on the same cable as 3 pins are exposed in the Filament sensor connector, two I2C, one analog pin. That would have been the perfect setup for me.
What I find disappointing is the fact that they gave up on a feature that was even a selling point for the MK3(NOT S) and that was jamming detection. Maybe this changes with the MK4 and we could have a design incorporating both sensors.

wavexx, 2019-11-29:
The relevant PR has been merged. @sfeine @amherms are you able to build the FW and test the change?

mherms, 2019-11-30:
> The relevant PR has been merged. @sfeine @amherms are you able to build the FW and test the change?

Thanks! Yes, I should have time to test it tomorrow.

mherms, 2019-12-09:
> > The relevant PR has been merged. @sfeine @amherms are you able to build the FW and test the change?
> 
> Thanks! Yes, I should have time to test it tomorrow.

OK, so I turned the filament sensor back on in the official Prusa firmware v3.8.1 and I cannot get it to fail like I did before. I wanted to confirm that I still had the problem before trying the compiled FW from the merge. Is there a chance that the problem was already fixed in v3.8.1?

leptun, 2019-12-10:
@mherms I don’t think 3.8.1 has any fix. @wavexx has started working on a fix and if I’m not mistaken it’s been merged into the latest MK3 branch and it is scheduled for 3.9. I don’t recommend you use this alpha version yet as there are some serious bugs which need to be addressed this week.

As for triggering the issue, you need to print long lines and trigger a runout check at the same time. A huge first layer where you intentionally make it check for the runout might work. Try shining light into the sensor from the side, maybe you can confuse it this way.

ryanneufeld, 2020-01-17:
I was able to repro this on 3.8.0 and 3.8.1

wavexx, 2020-01-27:
3.9.0 beta should include the fix for this

closed 2023-10-10 by github-actions[bot]