prusa3d/Prusa-Firmware #2265: MBL consumes excessive CPU time (3.8.1-RC1)

vintagepc, 2019-10-11:
When in stealth mode the mesh bed levelling procedure is consuming excessive CPU cycles, to the point where it is throttling the heat bed PWM and interfering with menu/LCD responsiveness. I've done a number of prints in normal mode and they did not exhibit this problem.

The menu/display only responds when the printer is moving (e.g. to a new location or tapping to sense height. It stalls during the same time the printer itself stalls in between measurements.
The heatbed PWM can be observed doing the same thing - the intensity of the LED only changes when the printer is moving as well. 

I've made a video demonstrating this, will update when it finishes uploading and processing.

vintagepc, 2019-10-11:
Update: video is here:
https://vimeo.com/365878731

Pay particular attention to the headbed LED in the second half. It is normally persistently on at the medium intensity for this temperature setting, but because something is consuming so much processing time it actually blinks on and off.

leptun, 2019-10-12:
It is true that switching from stealth to normal and the other way around takes time, but it’s not that bad. Looking at the video, I can’t see any problem with the bed led. As for the menus, lcd_update is not happening during the probing as it takes a considerable amount of time and it would make measurements less accurate.

vintagepc, 2019-10-12:
This isn't about the actual switch, it's that MBL takes a *lot* longer in stealth mode and I don't recall it being this slow (specifically, the long waits between moving from point to point)

Your explanation makes sense for the menus, but I still think something's not quite right. Let me see if I can make a different video and show the issue better.

leptun, 2019-10-12:
It takes much longer because it switches to normal mode for the probes, then to stealth for movement, then back to normal for the probe and so on...
One solution would be to stay in normal mode during the whole MBL, but that would make it noisier. I think this is a duplicate issue, but can’t find right now the other one.

vintagepc, 2019-10-12:
Ah, I see, that makes more sense. 

If that mode change is something that's expected to pre-empt changes to the heatbed PWM then we can close this as a non-issue. I did not realize it was changing modes on the fly.

closed 2019-10-14 by michalxfanta