prusa3d/Prusa-Firmware #2267: PSU clicking noise after upgrade to 3.8.0 MK3

DylanBarber, 2019-10-12:
Just updated to (prusa3d_fw_MK3_3_8_0_2684) on my MK3 (Silver PSU). I am in the U.S. 
I had clicking long ago but it was fixed and never happened again. Just updated and now it is clicking even worse than it did before and it's not the heatbed this time either (Happens during a print). 
I can email my prusa slic3r config file if needed.
Thanks!

michalxfanta, 2019-10-14:
What is your previous version of firmware?

DylanBarber, 2019-10-14:
V3.7.1 (1.0.5)

devcodeio, 2019-10-21:
Im having the same clicking "problem", although it was clicking prior to this firmware as well.  Its not a real issue for me per say, i just bring it up because i thought this issue was taken care of in the 3.8.0 release.
MK3S , Silver PSU (Canada), running 3.8.0 and now 3.8.1

vintagepc, 2019-11-03:
I noticed in relation to this that it's highly dependent on the set temperature (more or less as expected). For example, now that ambient temperatures are colder, at 60c, it's practically silent, but at 65c (first layer temp), I can hear it across the room.

I suspect this will change during the year too as the ambient temperature changes and the power required to maintain the given temperature also changes.

It's still much better than it used to be and far from the loudest thing on the printer, so like the previous commenter, I'm just offering a data point rather than complaining.

closed 2023-10-10 by github-actions[bot]