prusa3d/Prusa-Firmware #2272: 3.8 firmware does not extrude in corners

kyjak, 2019-10-15:
The MK3 skips extruding in corners.
The filament is simply not fed, pauses, when the printer gets around the corner, it resumes.
This renders the printer pretty unusable.
The gcode can be found at https://uloz.to/file/XW3sSj4NuhyC/opi3mk3-duet-case-bottom-gcode?fbclid=IwAR1lJhuaDL8BSlCe5Fjv_k-pixRtpxp8Bkg2pu30tHqS9GllNqkG5qRwU8A
Picture of the behavior is at https://www.facebook.com/photo.php?fbid=10157222972660862&set=gm.2469280683342102&type=3&theater

leptun, 2019-10-15:
The picture link is broken. How far to the edge are you going? Depending on that we can determine if it’s a software issue or hardware one.
I’m suspecting it’s a hardware issue. The extruder motor cable might have some problems. Try restarting the printer and moving the extruder by hand to the edge where it stops extruding and see if it loads or it fails. The X motor should be unpowered (don’t move it from the menu). 
Please report back with preferably a video if it’s not a hardware issue.

kyjak, 2019-10-15:
[image]
Uploaded the picture here, is it accessible?
The print goes near xmin but not xmax, so there is still lot of space.

leptun, 2019-10-15:
Please try loading and unloading filament like I said. Try gently moving the bundle of wires from the extruder and see if you can cause it to stop loading or unloading (or show weird behavior for that matter). I’d also check the connection to the board. Try plugging and unplugging it a few times. Make sure that at the end it makes a satisfying click as it locks. It might not click, in that case just make sure it can’t be unplugged by accident.

leptun, 2019-10-15:
You are saying that this is a 3.8 issue. Does it also happen on older versions? Check this last as you might need to do some recalibration.

kyjak, 2019-10-15:
Will do the tests you asked, also will downgrade to 3.7 and rerun the print.

kyjak, 2019-10-15:
Ok, what I did:
- created a new STL 
- rotated the model on the bed
- checked extruder paths in simulation, there are no errors
- waited till the error occurs. As per picture, you can see even brim, which is wider than the model, is printed fine, with no interrruptions
- when errors started to occur, I tried to bend,pull,push the extruder cable repeatedly during print. I could not make it skip on any other path, just in the corners as it started to do by itself.

[image]

kyjak, 2019-10-16:
I let the printer finish the job,even with errors, to get a better idea, a bigger picture attached.
Will test older FW soon.
[image]

leptun, 2019-10-16:
I've looked into the gcode and found some weirdness. It uses M205 to set jerk very frequently (once for every segment). I think that this breaks something in the firmware. Try slicing the part in PrusaSlicer and see if that fixes the problem. I'll try printing the gcode myself in order to rule out the possibility of a hardware failure.

What material are you using? The temperatures are very weird. 
The bed is too hot for PLA. (max 60)
The hotend is too cold for most PETG. (in my experience, minimum 230)
The bed is too cold for ABS/ASA. (at the very least 100)

kyjak, 2019-10-17:
I will  disable jerking and reprint. I do PLA, MK3 is reporting temperature lower than actually is, so I have my temperature settings skewed. Prusa textured bed has a low region, is not perfectly flat and the mesh leveling does not fix this either, after trials, this bed temperature actually helps me cover that.

kyjak, 2019-10-18:
You were right, after manually removing the jerk settings, the print is ok. Now the question is, do you want to keep this bug open and fix firmware so it can handle jerk settings no matter how ofther they are, or shall I close this ticket?

kyjak, 2019-10-18:
I was thinking, why setting the jerk very often destroys the extrusion,and maybe the marlin fw has a reset sequence once this command comes and clears a variable-buffer that holds current speed/acceleration. Or, other option, it is a "heavy" command that stalls some routines for a while.
So this is probably better reopened in the Marlin project, not Prusa.

However, in case of CURA users (as me), if it is not fixed and they set different jerk settings for different parts of the print job, the prints will fail, without the knowledge why...

wavexx, 2019-10-20:
Setting jerk mid-print *should't*, at least in principle, cause issues. The only trouble I see is that setting an X/Y jerk of 30 if above what you can expect from an MK3, but more in line with what you can do with a bowden style printer.

closed 2023-10-10 by github-actions[bot]