prusa3d/Prusa-Firmware #2276: Initial extrusion

mkh595, 2019-10-16:
MK3S: 3.8.0-2684
MMU2: 1.0.6-372

When print starts, nozzle moves to left front corner and extrude without move for 12 seconds. This create blob which size depends if filament was already in the nozzle or just moved from MMU2.

[image]

Then moves right and back to the start position, nozzle catch this blob

[image]

and drop it somewhere in the print area. 

[image]

Another example

[image]
[image]

If I remember well, before upgrade to MMU2 there was initial extrusion shorter always with move without return back. 
Is there reason why nozzle is not moving? If has to create such blob, can be the move back shorter (e.g. 10mm less) to not hit this blob?
Issue is mostly on Powder coated sheet because adhesion is not so good like on smooth PEI.

vintagepc, 2019-10-31:
This sounds like a duplicate of #2069 

Note you can get this if your sensors (IR/FINDA) are not calibrated correctly, if the IR continues to report zero, the MMU will force-feed your extruder and make a blob until the IR sensor happens to trip.

mkh595, 2019-11-13:
It is not issue of IR sensor, I have LED indicator, IR sensor works corectly.

[image]

leptun, 2019-11-13:
@mkh595 is this a single color MMU print?

mkh595, 2019-11-15:
yes, single color MMU. 

Purge line is created by 'Beginning Gcode' in PrusaSlicer delivered in '...MK3S MMU2S single' profile. I modified it to return 10mm less to not hit blob.
Size of blob depends if filament was unloaded from previous print or was already in nozzle, e.g. when previous print was cancelled by panic button.

```
; purge line
G1 X55.0 F2000.0
G1 Z0.3 F1000.0
G92 E0.0
G1 X240.0 E25.0 F2200.0
G1 Y-2.0 F1000.0
G1 X55.0 E25 F1400.0
G1 Z0.20 F1000.0
;G1 X5.0 E4.0 F1000.0    <= Do not return to start position
G1 X15.0 E4.0 F1000.0 
```

JohnnyDeer, 2021-02-03:
Duplicate of #2069, I´m closing.

closed 2021-02-03 by JohnnyDeer