prusa3d/Prusa-Firmware #2280: Frequent error 'Some problem encountered, Z-leveling enforced' - Rel 3.8.0 - Mk2.5S

joris57, 2019-10-18:
When mesh calibration runs, I frequently get an 'Some problem encountered, Z-leveling enforced' error. Moving back to rel. 3.7.1 the problem goes away. Printer prints fine.

Maybe the limits for this check are set too tight in rel 3.8.0?

leptun, 2019-10-18:
Duplicate of #1651. A fix is on it's way. If everything goes according to plan, it should be fixed in 3.8.2.

joris57, 2019-10-20:
Thanks for the info, but it is worth noting that #1651 is about an issue in 3.7.0-RC1, where the issue I encounter does not manifest itself (in my set-up) when I migrate back to rel. 3.7.1

Kachidoki2807, 2019-10-22:
@joris57, do you mean that your issue doesn't disappear when probing with heatbed off?
The #1651 thread mixup some different issues, but it solves at least one. 
Could you be so kind to try out the firmware posted by @leptun [here](https://github.com/prusa3d/Prusa-Firmware/issues/1651#issuecomment-538714471)?
This one solved my 'Z-leveling enforced' issue, related to the heatbed PWM.

If it doesn't work better for you, then it will confirm that there are at least two issues giving this error.

joris57, 2019-11-20:
Sorry for he delay in replying. I tried the firmware referred to by @Kachidoki2807 and I am still experiencing the same issue.

madchiller, 2020-02-07:
Any updates on this... since updating to 3.8.1 this issue occurs frequently, and I starting to get me?  Any RC of 3.8.2?  I’m down to test...  thanks guys!

leptun, 2020-02-08:
@madchiller the latest MK3 branch has the mention linked above merged. It should greatly improve the number of rechecks.
You need to compile the fw by yourself. It is not present in any releases as of now. It will be present in the upcoming 3.9.0-RC1.

Deneteus, 2020-02-13:
@leptun Is this really only an issue because the Pinda V2 is not shielded? Mine started doing it after upgrading to the last beta but it wasn't doing it before that. It's only failing near the back left side where the power comes into the heat bed.

[image]

leptun, 2020-02-13:
I also have a retry in one of the points on the last row. In the latest beta we preserve the bed state (on or off) during the probing. Doing this seemed to reduce the amount of retries in my case, but apparently that is not universal.
I'm trying to improve the accuracy even more. I've noticed that the extruder has quite some wobbling right after it stops on the point it wants to measure. Maybe That is contributing in a negative way to the measurement. Can you please give a try to the changes here? https://github.com/leptun/Prusa-Firmware/tree/MK3_bed_pwm
It's just a quick attempt to figure out if this has any positive effect or not.

Deneteus, 2020-02-13:
I think there is some sort of delay during sensor reads because I had the sensors state readout screen up while it was doing 5 tries per point using 7x7 and I noticed it was barely able to trigger it would do trigger 2-3 out of 5 but never 5 out of 5. I got that wobbling you were talking about but its more like the reads are too quick. I had already tried resetting the Pinda height and it failed over and over for like an hour.  So then I tried using a piece of aluminum tape over the top cap for the bed wiring and it was able to complete that last point. I have gotten it to work three times now. I have to finish setting up my Arduino IDE  settings so I can test builds. I wish there was  a nightly builds for the firmware.

CunningLogic, 2020-04-21:
This weekend I updated my mk3 to a mk3s, and from 2yr old firmware to 3.8.1. I went from a rock solid printer to an unstable and unusable printer. I'm hitting "some problem encountered, z-leveling enforced" all the time. It fails to return to home by 1 to 2 inches, so when it does leveling it crashes into the right side of the x axis. Pinda checks out fine, belts are spot on, ive factory reset it, ive done dozens of calibrations. Nothing seems to fix it. Spent about an hour with prusa support. Thought it was solved, first print after disconnecting ... and it's back. Any input or suggestions not already laid out in this thread?

floridaservices, 2020-04-21:
> This weekend I updated my mk3 to a mk3s, and from 2yr old firmware to 3.8.1. I went from a rock solid printer to an unstable and unusable printer. I'm hitting "some problem encountered, z-leveling [...]

reset printer to defaults and run through xyz again.  If it won't get through xyz check your bearings for binding.

CunningLogic, 2020-04-21:
>>reset printer to defaults and run through xyz again. If it won't get through xyz check your bearings for binding.

I've reset it and it continues. What is the best way to check and solve that bearing binding?

floridaservices, 2020-04-21:
> > > reset printer to defaults and run through xyz again. If it won't get through xyz check your bearings for binding.
> 
> I've reset it and it continues. What is the best way to check and solve that bearing binding?

This is what i have done (it fixed my xyz calibration)- loosen the belt to where you can slide the axis back and forth without the motor getting in the way, it will be easier to feel issues this way without the motor engaged.  The axis should slide nice and smooth, no rough spots.  Not necessarily quiet, just smooth.  Go all the way from end to end.  If you find them to be rough you can try greasing them with superlube, (there's a nice print for a grease nozzle that screws right onto the tube of grease).  Grease may help marginal bearings but I would invest in some decent bearings, it really does make a difference to how well the printer runs.  You don't need Misumi necessarily, just avoid the no name "ebay" LM8uus.  Get something where you can actually pull up a website for the manufacturer and look at specs and that will probably work fine, we are not building a rocket to go to the moon here.

You should also try loosening the u-bolts a 1/8 turn(really that little).  Overtightening the u-bolts is a pretty well known assembly error that just about everyone makes.  It's way too easy to overtighten the u-bolts and this will cause binding as well.  The way I have learned to do it - tighten the u-bolt equally to just touch the bearing, then try to move the U-bolt at the bottom of the bearing back and forth, just light pressure.  If you can move it then tighten it up only enough to where you cannot move it any more.  Any more and it's too tight and will cause the bearing to drag more than designed, which causes binding.   

Sorry for the lengthy post, these are just a couple things that I learned from doing -  and they fixed my calibration issues. I hope this helps you

CunningLogic, 2020-04-21:
>

Thank you. If memory serves me right, there are no ubolts holding the bearings on the x axis, which is the one giving me issue here?

leptun, 2020-04-21:
@CunningLogic The U-bolts are for the Y axis.
The same overtightening issue can happen on the X axis as well. In this case, it's the plastic part that is covering the bearings that is too tight. On the MK3S it's held in place with 4 screws. Try releasing a bit all of the screws and see if it gets better. Also, make sure you haven't overtightened the zipties that are holding the top bearings.

floridaservices, 2020-04-21:
@CunningLogic Right I should have said the u-bolts were y axis only, y was where my problems were.  What Leptun said regarding x should help.  Same trick with loosening the belt will apply, should allow you to slide the axis by hand and feel for issues easier.

CunningLogic, 2020-04-21:
> @CunningLogic The U-bolts are for the Y axis.
> The same overtightening issue can happen on the X axis as well. In this case, it's the plastic part that is covering the bearings that is too tight. On [...]

Belts are spot on for tensions, y axis hasn't been touched. It doesn't appear to be a y axis issue, appears that it does not go left enough to find home, so it crashes into right side when leveling

floridaservices, 2020-04-21:
have you checked to see if you can move the x axis to engage the min stop by hand?  Something as simple as a zip tie out of place on the cable management can interfere with the homing process. It's a pretty tight setup so everything needs to be exactly how the instructions show it.

CunningLogic, 2020-04-21:
> have you checked to see if you can move the x axis to engage the min stop by hand? Something as simple as a zip tie out of place on the cable management can interfere with the homing process. It's a pretty tight setup so everything needs to be exactly how the instructions show it.

Sorry, not sure what min stop is. I've taken off and redone all the zipties already, ive taken the extruder darn near back to pieces. I will likely take it all apart again and rebuild, but first im going to roll back to an earlier firmware and see if that changes anything.

floridaservices, 2020-04-21:
Forgot we are working with mk3s sorry.  Have you checked that x slides perfectly like i described?  If it's hanging up TMC driver can detect load and determine "home" at that spot.

CunningLogic, 2020-04-21:
> Forgot we are working with mk3s sorry. Have you checked that x slides perfectly like i described? If it's hanging up TMC driver can detect load and determine "home" at that spot.

I will shortly, waiting on my workshop to warm up.

CunningLogic, 2020-04-21:
Took it all down, rotated bearings a lot, put back together and so far so good, however I saw this yesterday as well.

natewaddoups, 2022-03-14:
There are a couple threads on the Prusa forum about this.

I ran into this error message yesterday with my MK3S, and tried all the recommended fixes with no change in behavior. 

And then I reverted to firmware 3.9.1, because the release note about firmware calibration being rewritten in 3.9.4 made me suspicous. With no mechanical changes, just the firmware downgrade, the Z calibration succeeded on the first try.

I'm guessing that the newer calibration code expects the pinda to be more accurate than it is. (More accurate than mine is, anyway.) 

I notice that there's a xyzcal_meassure_pinda_hysterezis function in the code that isn't being used yet, so I wonder if that might help.

floridaservices, 2022-03-14:
I would be interested to know where in the home circle that your Pinda is landing.  My guess is it is on the outer fringe of the circle and the older firmware is more forgiving of that.  Just a guess though.  The newer z calibration is better in my opinion because it is not as likely to tear up the heat bed as the old routines.

natewaddoups, 2022-07-04:
@floridaservices How do I find out "where in the home circle [my] Pinda is landing?" And how do I adjust it, if it's not centered?

I thought this was fixed after my last set of prints in March, but I tried to print again today and it had to go through the whole test/retest thing 3-4 times before it passed.

Thanks!

Prusa-Support, 2022-07-06:
Hello! Sorry for intervening so late on this thread. @natewaddoups, I believe @floridaservices referred to "where the PINDA ends up" after the "auto-home" procedure (>menu >calibration >auto-home). If it wouldn't end up on the "circle", then it would likely point to mechanical issues - belts, misaligned pulley, etc. We can expand on this aspect if needed. @floridaservices did you mean it this way, right? 

I don't have high suspects of mechanical issues (if it lands slightly outside of the circle, it would be more about fine-tuning), but since the issue depends on firmware version, it could have to do with PINDA probe position - can you please send me pictures of PINDA+nozzle from the front of the printer, to compare their relative position? Like this: https://cdn.help.prusa3d.com/wp-content/uploads/2868fe8e712a2014_painted.jpeg. In case feel free to move the conversation with me info@prusa3d.com mentioning me in the subject ("attn: Alessandro"), we can review in detail case by case. 

@joris57 I am sorry that the offered support hasn't been consistent - unfortunately, on GitHub we have to recognize it hasn't been consistent on all threads so far, and working on an improvement on this now - but we as technical support have always been there on our official channels (email: info@prusa3d.com, and LiveChat 24/7). I see this negative experience led to abandoning your Prusa printer, but in case of anything to review for any future cases, feel free to contact me personally info@prusa3d.com mentioning me, we can expand on anything. 

Alessandro Pantaleo
Prusa Research

closed 2023-10-10 by github-actions[bot]