prusa3d/Prusa-Firmware #2282: 3.8.0 firmware causing X-axis motor to constantly crash into x-home

makermelissa, 2019-10-19:
I have experienced this a number of times and it seems this is how it usually happens.
1. Print something after turning on printer
2. Go to print something else
3. The hotend does not heat up and it remains in a "heating..." state
4. Go to Stop Print
5. Extruder goes crashing into side

Printer: Prusa MK3/S with MMU2S
Printing in Single Filament mode
Firmware version 3.8.0

makermelissa, 2019-10-19:
Here's a picture of the heating issue before the crash:
[image]

I knew once I saw this the crash would definitely happen.

makermelissa, 2019-10-19:
Oh yeah, one other thing is I didn't seem to have this problem until I put the gcode files a couple levels deep inside folders.

leptun, 2019-10-19:
Please try using 3.8.1-RC1. The problem might be fixed there.

makermelissa, 2019-10-19:
Ok, I'll give it a go.

leptun, 2019-10-19:
Out of curiosity, did you take the SD card out between prints? If yes, then that’s the exact issue I fixed recently and should be present in 3.8.1-RC1.

makermelissa, 2019-10-19:
Yes I did.

tibahut, 2019-10-19:
I had the same issue with SD card removal and print stop combo.
I can confirm the issue is fixed with the 3.8.1-RC1.

Thanks @leptun

closed 2019-10-24 by michalxfanta