prusa3d/Prusa-Firmware #2284: Feature Request: Cancel Individual Objects via M-command and LCD menu

3d-gussner, 2019-10-21:
@paukstelis Introduced Cancel Individual Objects at MRRF and now a newer version at ERRF2019.

This feature has been added to PrusaSlicer / Slic3rPE via a PR from @supermerill https://github.com/prusa3d/PrusaSlicer/pull/1092 and finally https://github.com/prusa3d/PrusaSlicer/commit/c82d346c1a5b119aeb8c5176f832ccf19e7dbd9f

@MarlinFirmware Implemented a new version with M-commands and LCD menu support few days ago https://github.com/MarlinFirmware/Marlin/pull/15590

@paukstelis also created a new PR https://github.com/prusa3d/PrusaSlicer/pull/3086 to add this new functions to the existing PrusaSlicer implementation. 

As the PrusaSlicer supports this feature, wouldn't it be great if the Prusa Firmware would it also support this?

androandra, 2020-08-26:
Bump. This would be nice to have.

sslupsky, 2020-10-13:
@michalxfanta Yes.  Please add this feature.

From time to time when printing a number the same or different instances, a defect may arise with one of the instances. If the user is watching the printer at the time the defect occurs the print can be paused and most of the time the defect can be fixed and printing can continue. For long prints however, it is not likely the user is going to attending the printer continuously during the many hours it takes to complete the print.

The user may inspect the progress of the print from time to time to see how it is progressing. If a defect occurs the print can be paused and an attemp can be made to fix the problem. The more time a print defect is allowed to progress the more likely the defect will be catastrophic to the remaining instances being printed. Either because the defect starts to cause defects in nearby instances or because the print head crashes. In such cases is becomes nearly impossible to save the remaining instances notwithstanding that the remainder of the print is intact and it is desirable to continue printing the good parts. In some situations, the defective part could simply be removed from the print bed. However, removing such defective part removes support for the remaining layers and therefore the printer prints spaghetti when it resumes printing.

Thus, if printing the defective instance could be canceled, the printing of the remaining instances could continue. Thus, this feature request is to add an option to cancel one or more instances currently being printed and continue printing the remaining instances.  This is quite important for a production workflow.

FEsmondeWhite, 2022-01-03:
This feature would greatly improve printability of longer-running multiparty prints. For example, I currently have a 22-hour 2-object print running right now. The parts are both quite tall, and the print is about 10 hours in. One object is experiencing minor issues, with some defects visible. I would love to be able to cancel printing of the defective object (#2), while allowing the object (#1, which is fine) to continue printing (and printing faster as there would only be a single print object remaining).

Likewise, if I could cancel specific object printing, the same gcode could be run again once the object #1 print is complete, and object #1 could be cancelled at the start off the next print, allowing object #2 to print without going back and slicing object #2 separately.

It sounds like this feature is already implemented, and would mostly involve integration and testing.

amoose136, 2022-06-03:
Particularly would be nice if this was a feature on the XL and Mini because they have color LCDs so PrusaSlicer could include an overhead image along with the angled thumbnail which would allow for canceling the object without pausing and moving the nozzle around to point at the object you want to cancel. Also on the XL you're more likely to be printing a bunch of a part due to the larger print volume.

sslupsky, 2022-06-03:
For the Mk3's character LCD I think the UI could simply use the instance number provided in PrusaSlicer if there is a way for the firmware to correlate an instance number with the instance's g-code.

FEsmondeWhite, 2022-06-03:
Agreed.

I think that scrolling the model name from Prusa slicer, with the instance
number, and allowing cancelling that object print would be wonderful.

I think it would be fairly easy for a user to pull up the prusaslicer
project and determine which model/instance needs to be cancelled.

It wouldn't need to be very complicated, as the code for each object layer
is already split in gcode, it would require the firmware to look for and
skip the specified labeled gcode sections.

paukstelis, 2022-06-03:
There are likely going to be limitations to storing object names in memory for use on the display. It would certainly be possible to have 50-100 objects on the build plate. Storing those would be problematic.  There are also some things that need to be worked out on the slicer side (sequential printing, start/stop tags relative to wipe positions).

closed 2023-10-10 by github-actions[bot]

jkavalik, 2023-11-25:
@Prusa-Support Did this get closed by error or overzealous bot?