prusa3d/Prusa-Firmware #2288: Firmware 3.8.1-RC1

Silverslim, 2019-10-22:
Just had a power failure ... No power panic function!

michalxfanta, 2019-11-08:
Please try the latest fw 3.8.1

leptun, 2020-01-26:
Were you printing from the SD or from USB/raspberryPI? if it's the latter, unfortunately the PowerPanic is not compatible with that. Only SD prints are recoverable.

Prusa-Support, 2022-06-14:
Hello! Sorry for intervening so late on this thread. Is the issue still actual? Was the print via SD or via USB, RPi? In case the issue is still there, can you please let me know via email info@prusa3d.com mentioning me? Closing the issue for now, but in case the issue is still there I can look into it in more detail via email. 

Alessandro Pantaleo
Prusa Research

closed 2022-06-14 by Prusa-Support