prusa3d/Prusa-Firmware #2289: Prusa 3.8.0 i3 MK3 Print Freeze

Foxtrek64, 2019-10-23:
Hello,

I had recently flashed to 3.8.0. After running one print successfully using a new filament (gold PLA), I had gone back to start a second print.

Selecting the file from the SD card, the printer began to pre-heat. It was at this point I noticed that I had forgotten to change filaments (red PLA). So, going to the main menu, I selected the unload filament option, thinking it would simply cancel the print.

Filament unload preheated the printer for PLA as I'd selected, then said to press the dial to unload the filament. Pressing the dial kicked me back to the main menu and the unload process failed to start.

I tried this a few more times to no avail. So, I felt it best to restart the printer. I turned the printer off using the power switch, waited for a complete shutdown, gave it approximately ten additional seconds, then turned the printer back on. It immediately went into print recovery mode and began preheating. The printhead was set to pre-heat to 167 and the bed was set to preheat to 80. So I let it preheat to see what happened. It heated to 167 and 80, respectively, and then sat there reporting that it was still heating.

At this point, I tried to cancel the print to find that there was no cancel option - it was merely stuck preheating. Attempting to unload also continued to exhibit the same behavior.

Rebooting with the power switch and the front-panel reset button did not have any effect. It would simply go immediately into print recovery mode and start preheating to 167 and 80 again. Next I tried the reset switch in the back. Same behavior.

Disabling recovery mode and restarting made it prompt whether to recover. No was selected and it appeared to return to normal operation, though this is where I stopped.

As a **temporary work-around**, I have downgraded the firmware to 3.7.1. After recalibrating, and clicking past the firmware update prompt, printing is operating as expected.

michalxfanta, 2019-11-08:
Please try the latest firmware 3.8.1

Dave5cents, 2019-12-21:
This behavior is still present in 3.8.1.

closed 2023-10-10 by github-actions[bot]