prusa3d/Prusa-Firmware #2290: 3.8.0 MK3s i3 potential memory corruption issue

XanTheDragon, 2019-10-23:
I noticed some incredibly odd and quite negative behavior on my MK3s i3 printer upon loading a model's gcode. This gcode used a slightly modified PLA preset which featured a slightly different extruder and bed temperature due to being a third party filament. 

**I am personally speculating that this is memory corruption given the nature of the values presented, but I am not 100% certain this is the cause.**

The issues noticed were as follows:
- Extruder attempted to heat to 0C (target temperature was 0). Heatbed temperature was not affected and heated properly. Extruder temperature probes were properly attached and the component is not faulty.
- Z axis offset was reported, in what numbers could be displayed on the LCD panel, to be 5.3 kilometers high (Yes, approx 5,300,000mm. I don't remember the exact digits). I do not know if the underlying value was larger than this and simply not shown due to the value exceeding the screen buffer. The PINDA sensor is not faulty and all testing said that it was fine, and the cables are not loose.
- Y axis offset was presumably malformed as well. Upon pressing the emergency stop button when the Z axis reported this insane value, the Y axis began to move back and did not stop moving back when it hit the physical limit of the printer. I flipped the power switch to avoid damaging the components, since it began damaging the belt due to the stepper motor continuing to rotate even after the bed could no longer move. No critical damage was done and the printer is still functional.

This issue is quite dodgy to reproduce and I can't say I'm 100% certain of what causes it. The only immediate known factors are that I am using Prusa Slicer v2.1.0 win64-201909160915, and my PLA preset had a custom temperature value of 230C for the first layer and 225C for the second layer. I attempted to print three times and this occurred in two of the three prints. The third print was successful, so I allowed it to finish. The firmware was downgraded shortly afterwards. This issue does not occur on the older firmware.

Issue #2275 may be reporting a similar problem (memory corruption), however the original poster did not include much information and it's not possible to discern if it is indeed a related issue. Contrary to #2275, I did not notice any LCD panel text corruption.

I've attached this custom PLA profile so that it may be cross-checked, but I have some reason to doubt that it is the cause of this problem (namely, changing temperature values should not cause such a catastrophic change anywhere else in the printer, such as the Y/Z axis faults).
[Metallic PLA.zip](https://github.com/prusa3d/Prusa-Firmware/files/3759427/Metallic.PLA.zip)

###### n.b. This is the same printer as #2289 -- We share a printer. I mention this because the information could potentially be of value when investigating the reported issues.

leptun, 2019-10-23:
Did you take out the sd card between prints? Was the file in a/multiple folder(s)? If yes, then the corruption issue might be fixed in 3.8.1-RC1.
See #2163 #2196 #2237 with fix #2240 

The #2275 lcd corruption is unrelated to this. There is only some language corruption in that menu with that language that should affect everyone using that version, but shouldn’t corrupt memory.

XanTheDragon, 2019-10-23:
"Was the file in a/multiple folder(s)?"

This is the case that applied to me -- My model was pretty far down a folder structure.

leptun, 2019-10-25:
I suppose the issue can be closed, then. 3.8.1 is now released and it includes the fix just like the RC.

closed 2019-11-08 by michalxfanta