prusa3d/Prusa-Firmware #2294: 3.8.0 Firmware on MK3S - XYZ Calibration text all garbled

ctabrizi, 2019-10-24:
The calibration still seems to work but all the instructions broke. Uploaded firmware via USB cable from PrusaSlicer 2.0.0+. See photo below:

[image]

leptun, 2019-10-25:
Can you please provide more details about the issue?

Were you using a custom firmware?
Were you using English or another language.
Does it always break? Can you reproduce the issue again?
Does downgrading fix the issue?

prusaprinter, 2019-10-26:
i have the same problem but if i just let it run then it normal fixes the issue its self

Kachidoki2807, 2019-10-29:
Was it when you put down the steel sheet?

I got sometime, not often but it happens, the screen becoming blank when I put the steel sheet. On the next refresh the display come back to a normal status. I also already got this weird display, but I cannot remember precisely the conditions.
I suspect it is when I put the sheet too quickly or when the magnet are stronger than my fingers ans the sheet slip out. :-) This maybe produce an EMC pulse that break the communication with the display through the ribbon cables.

vintagepc, 2019-10-31:
The display is somewhat subject to static electricity, it has been known to get corrupted or even damaged from static zaps to the bed or display module. I've definitely had this happen a few times last winter when the air was very dry and at some point it damaged my LCD module. 

The PSU is grounded, so I suggest touching its metal casing before touching the printer or heat bed when the air is dry and static shocks likely.

closed 2023-10-10 by github-actions[bot]