prusa3d/Prusa-Firmware #2295: Feature Request: Don't heat the bed when using the autoload/unload filament menu

Josh3D, 2019-10-25:
The current way of changing filaments is:

- Select autoload/unload filament menu

- The printer will then ask you to select a material that you are either loading or unloading 

- The printer then will preheat the Hotend AND heat bed to those temperatures

I propose the following when selecting the autoload/unload menu, and the temperature selection pops up that it only will heat the Hotend and NOT the heat bed.

Some benefits of this are: 

- Not accidentally burning yourself on the heat bed when you touch it changing filament or taking off the purged plastic that has dropped down on the bed

- After removing the filament you had in your printer previously, loaded in the filament for your next print, the bed would then still be room temperature so you can wipe it down with IPA effectively and then be able to start your next print right away.

JKinch03, 2019-11-14:
For MMU2 users, you could also 
-Add this feature to the load/unload to nozzle
-Change the "Eject Filament" feature so that it checks if the PINDA probe and the IR sensor are triggered, and if neither are triggered when the eject function is issued, then you wouldn't need to preheat the nozzle or heat bed to eject the filament. If the IR sensor **OR** the PINDA are triggered, then it'll still preheat the nozzle for safety measures.

Prusa-Support, 2022-05-19:
Duplicate of #422

closed 2022-05-19 by Prusa-Support