prusa3d/Prusa-Firmware #2296: Prusa MK3S bed calibration point not found

prusaprinter, 2019-10-26:
My Prusa Mk3s that I just got will not finish the bed calibration points, every time it says "bed calibration point not found". I have tested the P.I.N.D.A. probe and tensioned the belts to the perfect point but I still am not getting it to finish. I have tried to follow every comment online but none of them that I could find describe my problem. Every time it took 30 seconds to look for the calibration point but ends up just moving past the calibration point with the P.I.N.D.A. sensor not being close to the dashed circle.
	Please help I have been trying to fix it for about a week now and nothing will work.

leptun, 2019-10-26:
A video of the calibration might be helpful. Does the printer complete selftest?

prusaprinter, 2019-10-26:
yes

prusaprinter, 2019-10-26:
i tried to watch videos of the problem but i couldn't find any

leptun, 2019-10-26:
I'd like you make a video and upload it to something like google drive. I want to see how it does the calibration. What firmware version are you using?

prusaprinter, 2019-10-26:
i do not have the printer now but i might be able to get a video of it by wed

prusaprinter, 2019-10-26:
also the firmware that i am using is 7.1.2 i think

leptun, 2019-10-26:
That version number never existed. Try re-flashing 3.8.1 firmware once you get the machine. https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.8.1

You have to use PrusaSlicer to upload the firmware to the machine.

prusaprinter, 2019-10-26:
i will try to do that on wed to but i found the firmware
https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.7.2

3d-gussner, 2019-10-26:
You can find the latest release https://github.com/prusa3d/Prusa-Firmware/releases/latest :-)

Tanischacat2009, 2019-10-26:
Double check your PINDA probe height, reset your printer and try again.  This might help as I just updated firmware from 3.8.0 to 3.8.1 on my MK3s and had to try 3 times to get it to work, but it did work.  Adjusting the PINDA probe height corrected the error message that I was getting.

prusaprinter, 2019-10-26:
i will trey to do that soon

acidmang, 2019-10-29:
I'm having the same problem...  I did not run into it with 3.8.1 RC-1 but the 3.8.1 release seems to be random.  Still need a video?

michalxfanta, 2019-11-08:
Please send us a video.

prusaprinter, 2019-11-08:
I found that if I go to settings and then to move axis, then I would put the x and y to 0 the I would disable steppers and move the axis there. That would calibrate so then I would start a calibration and that worked

closed 2019-11-08 by prusaprinter