prusa3d/Prusa-Firmware #2298: MMU2S single mode -  got error when running out of filament

ThaliaFromPrussia, 2019-10-28:
Hi there,
I use my MMU2S in single mode, printing with PLA.
Somewhere mid print it stops, gets a weird message on the LCD an there is no way to recover the print. Unluckily I did not save a pic or the error messages. But just a few minutes ago, the printer did the same thing, as it ran out of filament. Normally, it whould ask for attention, but it just gave me this:
[image]

This is what octoprint gave me in the terminal:
````
Recv: SILENT MODE: Percent done: 49; print time remaining in mins: 625
Recv: ok
Send: N288415 G1 X222.358 Y18.671 E1.20482*104
Recv: ok T:215.4 /215.0 B:60.0 /60.0 T0:215.4 /215.0 @:99 B@:20 P:32.6 A:44.0
Send: N288416 G1 X224.294 Y27.435 E0.30378*96
Recv: ok
Send: N288417 G1 X224.373 Y27.696 E0.00924*98
Recv: ok
Send: N288418 G1 X224.562 Y28.102 E0.01516*98
Recv: ok
Send: N288419 G1 X224.792 Y28.415 E0.01315*104
Recv: ok
Send: N288420 M105*19
Recv: fsensor_stop_and_save_print
Recv: echo:Enqueing to the front: "PRUSA fsensor_recover"
Recv: echo: 3.8.0-2684
Recv: echo: Last Updated: Sep  6 2019 19:51:14 | Author: (none, default config)
Recv: Compiled: Sep  6 2019
Recv: echo: Free Memory: 2116  PlannerBufferBytes: 1392
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect ENABLED!
Recv: FSensor ENABLED
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N288421 M105*18
Recv: echo:SD init fail
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1
Printer requested line 1 but no sufficient history is available, can't resend
Changing monitoring state from "Printing" to "Error: Printer requested line 1 but no sufficient history is available, can't resend"
Send: M112
Send: N288422 M112*23
Send: N288423 M104 T0 S0*22
Send: N288424 M140 S0*85
Changing monitoring state from "Error: Printer requested line 1 but no sufficient history is available, can't resend" to "Offline (Error: Printer requested line 1 but no sufficient history is available, can't resend)"
Connection closed, closing down monitor
````
It happens via SD-Card or when printing with Octoprint.

ThaliaFromPrussia, 2019-10-28:
Flashing new firmware right now, I hope this helps.

leptun, 2019-10-29:
Please make sure to flash the firmware using PrusaSlicer. If you use anything else, then the languages will not be updated from the previous version snd everything might seem to work, until you get to one of the messages that did not exist in the old version.
How I figured this out? You have _SEC_LANG on the display. This is one of the marks for the firmware during compiling in order to identify where the languages area is located in memory. In you case the text was corrupted and it got to the tag and printed it.
Btw, did you reset the printer or did it reset by itself? From the gcode log it is unclear.

gudnimg, 2023-08-02:
Closing as this is stale. If there is still an issue, we'd like a new report with newest firmware (3.13.0 at the time of writing).

closed 2023-08-02 by gudnimg