prusa3d/Prusa-Firmware #2300: Live Z-index disappears during print FM: 3.8.1-2869

theuvibe, 2019-10-30:
If you're in the tune menu and long press for z-index, it flashes. I am then not able to get the live z-index for the duration of the print. 

#FM:  #3.8.1-2869

[image]

[longpress.mov.zip](https://github.com/prusa3d/Prusa-Firmware/files/3789937/longpress.mov.zip)

BeldrothTheGold, 2019-10-31:
A change was made to the way the adjust Z/ live adjust Z work in the 3.8.1 release.
This was to resolve several bugs the previous implementation caused.

Pulled this from the 3.8.1 release notes:

"New Live Adjust Z shortcut
Right after the mesh bed levelling as soon as the print is started, the user can adjust from the menu the height between the tip of the nozzle and the steel sheet (Live Adjust Z) to ensure the filament is properly squished in the printing surface.

Now, there is a new shortcut to access this function. Long press the rotation knob to enter the adjustment menu. The function is available only during the print between 0-2 mm of height.

The Live Adjust Z shortcut is allowed in the following menus (during the print only):

Info screen
Main menu
Tune
Support
Move Z shortcut adjustment
To move the Z-axis up or down, the user can either choose this option from the menu or long-press the rotation knob. Because this shortcut is now shared with the Live Adjust Z, the Move Z is available only before the mesh bed leveling starts and after the print is finished.

Previously, it was also possible to use the shortcut during the print, this is now disabled. #2165 #2101

The Move Z shortcut is allowed in the following menus:

Info screen
Main menu
Preheat
Print from SD
Settings
Temperature
Select language
Support"

theuvibe, 2019-10-31:
> A change was made to the way the adjust Z/ live adjust Z work in the 3.8.1 release.
> This was to resolve several bugs the previous implementation caused.
>
> Pulled this from the 3.8.1 release notes: [...]

Hey Scott,

Once the live z disappears, it is not available or shown in any menu item at all.

Cheers

Rhys

BeldrothTheGold, 2019-10-31:
I believe part of the change was also disabling live z-height adjustment during printing once the print has gone past 2mm. Is that what you are seeing?

theuvibe, 2019-10-31:
Ah yes, that looks like what's happening. I didn't realize that was intentional, thanks!

3d-gussner, 2019-10-31:
@BeldrothTheGold Well done :-)
@theuvibe If your question/issue is answered, can you please close the issue?

Less open issues == better and faster response.

And people having a new issue should try to search also in closed issues, if that has been already solved.

theuvibe, 2019-10-31:
@3d-gussner yep, closing now :) thanks!

closed 2019-10-31 by theuvibe