prusa3d/Prusa-Firmware #2301: 3.8.0-3.8.1 Printing MK3 gcode with MMU2S reboots if no filament loaded.

eltoozero, 2019-10-30:
Applies to MMU2/S only.

Previously, the print would initialize normally: after calibrate as purge begins, extruder would park and user is prompted for which filament to load; in 3.8.0 and 3.8.1 the printer just reboots.

If filament is pre-loaded when printer starts, printing MK3 gcode will cause reboot, the machine must actually know which filament is loaded; F? status will cause reboot when printing MK3 gcode.

Workaround: Implicitly load filament to nozzle before printing MK3/S gcode.

I'm testing against the included Prusa_200 g-code.

BeldrothTheGold, 2019-10-31:
A change was made to the way the adjust Z/ live adjust Z work in the 3.8.1 release. 
This was to resolve several bugs the previous implementation caused. 

Pulled this from the 3.8.1 release notes: 

"New Live Adjust Z shortcut
Right after the mesh bed levelling as soon as the print is started, the user can adjust from the menu the height between the tip of the nozzle and the steel sheet (Live Adjust Z) to ensure the filament is properly squished in the printing surface.

Now, there is a new shortcut to access this function. Long press the rotation knob to enter the adjustment menu. The function is available only during the print between 0-2 mm of height.

The Live Adjust Z shortcut is allowed in the following menus (during the print only):

Info screen
Main menu
Tune
Support
Move Z shortcut adjustment
To move the Z-axis up or down, the user can either choose this option from the menu or long-press the rotation knob. Because this shortcut is now shared with the Live Adjust Z, the Move Z is available only before the mesh bed leveling starts and after the print is finished.

Previously, it was also possible to use the shortcut during the print, this is now disabled. #2165 #2101

The Move Z shortcut is allowed in the following menus:

Info screen
Main menu
Preheat
Print from SD
Settings
Temperature
Select language
Support"

BeldrothTheGold, 2019-10-31:
Disregard my previous comment. Was on wrong issue. Sorry.

vintagepc, 2019-10-31:
This sounds like it might be a duplicate of #1993, or at least have a shared cause. 

Check your IR sensor and FINDA, this is known to happen if the sensor states disagree.

eltoozero, 2019-10-31:
FINDA and IR Sensor are groovy, normal MMU2S printing and Single Printing are working fine; it's specifically starting MK3 gcode causing a reset.

vintagepc, 2019-10-31:
👍 Good to confirm those are all OK. I usually just leave my filament loaded if I'm doing a string of single-mode prints but now that you mention it, I think I do recall seeing this as new behaviour when I forgot to do that. 

There's a couple weird corner cases in the firmware like this that cause a hard reset (e.g. if the x or y homing fails, as well).

leptun, 2019-10-31:
@vintagepc the X Y homing reset? Never heard of it. Can you please tell me more about it? I’ve been able only to corrupt memory when homing, but that was under very specific circumstances and running well above the einsy’s cpu capabilities.

vintagepc, 2019-10-31:
> @vintagepc the X Y homing reset? Never heard of it. Can you please tell me more about it? I’ve been able only to corrupt memory when homing, but that was under very specific circumstances and running well above the einsy’s cpu capabilities.
@leptun 
I was just searching if there was an open issue on it to amend. Did not see one, so I just filed #2306

Mirarkitty, 2019-11-21:
Sounds like the opposite of #2115...

vintagepc, 2019-12-01:
I just had my printer reset on filament runout mid print. Was on slot 5 finishing up a spool of PETG. The printer froze when the FINDA tripped, and then soft reset (nozzle fan did not come on full strength like a hard reset). Fortunately I was only an hour or so into the print, and not a lot was lost, but seems like something's not right with the FINDA/IR combo in certain circumstances.

glowfishDE, 2019-12-12:
I have the same issue as stated in #2363 (which i have closed in favor to this thread).

Kelly12612, 2019-12-12:
This just happened to me this morning, then I saw an email about this issue.
I was having adhesion issues, so I re-ran the live-z calibration, at the end of that, the filament is retracted from the nozzle, which in itself is annoying because that was the filament I was going to print with.
I then started a print (from the prusa lack v2 downloaded g-code) in MMU2 Single mode, and the printer reset.  This happened several times, then I tried loading the filament to the nozzle just to make sure that was going to work, once I did that the print started normally.

PR #2466 (2020-03-26): #2464 fix runout for unknown filament (also fixes #1993, #2301)
This fixes the printer soft-resetting if a filament runout is detected while the current filament is unknown (e.g. user loaded to nozzle, and an event caused a power cycle (power panic during MMU print), or user turned off printer and continued printing with the same filament the next day). 

Because the current filament is unknown, we cannot reliably retract it and move on to the next one if spooljoin is enabled (theoretically we could try this and let the MMU try a retract/home/next filament, but that seems risky),  nor can we m [... cut]
closed 2020-03-26 by DRracer