prusa3d/Prusa-Firmware #2302: PINDA temperature drops down

samuelemarazzita, 2019-10-31:
Hi guys, I have a big problem: after the first layer print, my temperature drops down 15° (FIFTEEN!!) while the printer continues with the next levels. This makes the temperature (with PLA settings) drops under 180° and doesn't make the filament melt enough causing a missing-layer issue. I tried to calibrate the PINDA temperature many times but with no success. I attach screenshot of the temperature monitor and the Slic3r settings. Room temperature is stable at 23°. Anyway this may not be important because the temperature drops in a short time.

[image]

[image]

3d-gussner, 2019-10-31:
@samuelemarazzita What you see in Octoprint are the temperatures of the hotend and bed. The MK3/s has default PID tuning values for 210°C.
So if you see that kind of overshooting of the hotend at lower/higher temperatures it makes sense that you run PID tuning for the hotend.

I think that the naming of the hotend PID tuning and PINDA temp drift tuning are not optimal. They can be be miss understood.

- "PID Calibration" = Hotend PID tuning
- "Temp. calibration" is PINDAv2 temperature drift calibration 

Solution:
- If you print most times around 180-185°C use the LCD menu "Calibration -> PID calibration" and set the target temperature to 180°C. This will run the PID tuning and store it permanent into EEPROM.

- If you normally print with other temperatures DON'T use the LCD menu
  - Check in Octoprint via G-code "M503" which PID values you have for hotend "echo M301...", it should look something like this: 
```
echo:PID settings:
echo:   M301 P16.13 I1.16 D56.23
echo:PID heatbed settings:
echo:   M304 P126.13 I4.30 D924.76
``` 

In Octoprint run the gcode "M303 E0 S180", where S180 stands for target temperature of 180°C.
  - write down the P I D values
  - add to this "lower temp Filament" an additional "Start G-code" with the values "M301 Pxx.yy Ixx.yy Dxx.yy" you got from the step before and safe it as custom Filament. PrusaSlicer -> Filament Settings -> Custom G-Code -> Start G-code
  - This will overwrite the default hotend PID settings just for this print/filament. But if you use other filament the default in EEPROM stored values are used.

Hope that helps.

samuelemarazzita, 2019-10-31:
Thank you @3d-gussner but the problem is still there. I've done everything you told me and the situation is still the same: temperature drops down to 170° from a 185° first layer temperature to a 180°.

[image]

This is the Filament Start G-Code I've wrote:
`M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{elsif nozzle_diameter[0]==0.6}18{else}30{endif} ; Filament gcode
**M301 P16.10 I0.98 D66.11** ; PID temperature calibration`

Maybe the problem is on my Printer Start G-Code? I paste it here.
`; Last updated 20190405
M115 U3.5.1 ; tell printer latest fw version
; Set initial warmup temps
M104 S160 ; set extruder no-ooze temp
M140 S{max(first_layer_bed_temperature[0],50)}  ; set bed PINDA warmup temp
; Nozzle warmup before home to avoid driving hardened ooze into PEI bed
M109 ; wait for extruder no-ooze warmup temp before mesh bed leveling, cool hot PINDA
M300 S40 P10 ; chirp
G28 W ; home all without mesh bed level
; PINDA warmup routine
M300 S40 P10 ; chirp
G0 Z3; Raise nozzle before move
; Present bed for final cleaning
G0 X125 Y210 F10200; Move nozzle to PINDA warming position
G0 Z0.10 F10200; Lower nozzle to PINDA warming position
M860 S35 ; wait for PINDA temp to stabilize
G0 Z3; Raise nozzle before move
M300 S40 P10 ; chirp
G80 ; mesh bed leveling
G0 Z3; Raise nozzle to avoid denting bed while nozzle heats
M140 S[first_layer_bed_temperature] ; set bed final temp
; Final warmup routine
M104 S[first_layer_temperature] ; set extruder final temp
M109 S[first_layer_temperature] ; wait for extruder final temp
M190 S[first_layer_bed_temperature] ; wait for bed final temp
; Prime line routine
M300 S25 P10 ; chirp
G0 Z-3; Restore nozzle position
M83  ; extruder relative mode
M900 K0; Disable LA for prime line
G92 E0.0 ; reset extrusion distance
G1 Y-3.0 F1000.0 ; go outside print area
G1 E2 F1000 ; de-retract and push ooze
G1 X20.0 E6  F1000.0 ; fat 20mm intro line @ 0.30
G1 X60.0 E3.2  F1000.0 ; thin +40mm intro line @ 0.08
G1 X100.0 E6  F1000.0 ; fat +40mm intro line @ 0.15
G1 E-0.8 F2100; retract to avoid stringing
G1 X99.5 E0 F1000.0 ; -0.5mm wipe action to avoid string
G1 X110.0 E0 F1000.0 ; +10mm intro line @ 0.00
G1 E0.6 F1500; de-retract
G92 E0.0 ; reset extrusion distance
; Final print adjustments
; M92 E282.54 ; adjust extrusion ratio
;M221 S{if layer_height>=0.32}95{else}100{endif} ; compensate for large layer height`

3d-gussner, 2019-10-31:
@samuelemarazzita As @leptun mentioned above it could be the part cooling fan. 

Lets try to investive ...
Try in Octoprint:
1. Let the printer cool down
2. Set you found PID settings
   -  `M301 P16.10 I0.98 D66.11`
3. Set the target temperature of the hotend to 185°C
   - `M104 S185`
4. Wait until it reaches the the 185°C
5. Set the target temperature of the hotend to 180°C
   - `M104 S180`
6. Check if it overshoots to 170°C
7. If it doesn't overshoot, set the target temp to 185°C
   - `M104 S185`
8. Let's check if the part cooling fan causing the issue by turning it on to 100%
   - `M106 S255`
9. Check in the graph if something is showing a significant drop
10. If nothing significant happens. Turn FAN off, set temp to 180°C and FAN 100%
     - `M107`
     - `M104 S180`
     - `M106 S255`

If the parts fan shows again the the drop, we know that your hotend on that temperature is sensible to high fan speeds.

Solutions could be:
  - Get a silicone sock https://e3d-online.com/catalogsearch/result/?q=Silicone+socks but make sure that is mounted properly
  - Modify "Before layer change G-code" with some `{if....}` statements to slowly speed up the part cooling fan. But this isn't that easy.

samuelemarazzita, 2019-11-03:
Hi guys, yes, it's the fan. I already had an e3d sock, but months ago I took it off because it was not perfectly fitting the hotend and in many prints was falling down while printing! I'll try to put it and let you know. Thank you for now!

samuelemarazzita, 2019-11-03:
Here I am. The sock helps, I Had to change all the hotend because the old one was so dirty that the silicon was flipping away.

Now the temperature, when the fan starts, drops to 176° just for a 5 seconds and the filament doesn't clog.

Thank you all!

closed 2019-11-03 by samuelemarazzita

3d-gussner, 2019-11-04:
@samuelemarazzita Glad you found and solved the issue. Thanks for closing the github issue, makes it way easier to focus on "real" open issues.
If you have trouble with the E3D sock try to fix it with something, or checkout this video to make your own https://www.youtube.com/watch?v=O5UiXky85hA&t=124s