prusa3d/Prusa-Firmware #2303: Wrong target temperature for heating up after power failure MK3s+MMU2s

Heckw, 2019-10-31:
Hi there,
I encountered a firmware problem after experiencing a power failure. My firmware-version is 3.8.0-2684. As the power returned, my i3 MK3s with MMU2s resumed printing and started heating up again. The problem was that the target temperature for the nozzle was wrong. It was set to 14°C.
As a result the nozzle kept cooling down and I had to set the correct nozzle temperature manually. The bed target temperature was correct.
Best regards,
Frans

leptun, 2019-10-31:
What was the target temperature before the power panic?

Heckw, 2019-10-31:
270°C 

Best regards,
Frans 

 

Links:
------
[1]
https://github.com/prusa3d/Prusa-Firmware/issues/2303?email_source=notifications&amp;email_token=AL7WCMO2ILNBRIUXBLNPC6LQRLE7VA5CNFSM4JHJC3IKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECXR3QQ#issuecomment-548347330
[2]
https://github.com/

leptun, 2019-10-31:
@wavexx There appears that Power Panic saves the hotend and bed temperature as uint8_t. This becomes a problem for temps over 255deg. Do you want to make a pr with a fix or should I do it?

Heckw, 2019-10-31:
> @wavexx [1] There appears that Power Panic saves the hotend and bed temperature as uint8_t. This becomes a problem for temps over 255deg. Do you want to make a pr with a fix or should I do it?

Ok. Yes please make a pr. Thanks! 

Best regards,
Frans 

 

Links:
------
[1] https://github.com/wavexx
[2]
https://github.com/prusa3d/Prusa-Firmware/issues/2303?email_source=notifications&amp;email_token=AL7WCMMNMSGSVXEXE6PQOFLQRLGTFA5CNFSM4JHJC3IKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECXTHGI#issuecomment-548352921
[3]
https://github.com/

PR #2307 (2020-01-27): Save/restore hotend temperatures above 255C in powerpanic
Save/restore hotend temp using two separate bytes.

Fixes #2303
commit 5d885737 (2020-01-27): Save/restore hotend temperatures above 255C in PowerPanic

Use 2 bytes to store extruder temperature during UVLO.

Re-use the storage of EEPROM_UVLO_TINY_Z_MICROSTEPS which has been freed
by previous changes.

Fixes #2303
closed 2020-01-27 by DRracer