prusa3d/Prusa-Firmware #2306: Printer hard-resets if X/Y homing fails (at least 3.7.2 - current FW)

vintagepc, 2019-10-31:
When homing (G28) if X or Y fails to trip stallguard, the printer will hard-reset. 

By far the easiest way to make this happen is with 0.9 degree motors and belts that are too tight or too loose. You need the M350 mod to enable setting X/Y microsteps. 

Generally, the printer will register the first impact correctly, but the second will not register and it will continue running into the end-stop for a few seconds before hard-resetting.

I'll try to force it to happen and make a video tonight when I get home.
There is some anecdotal discussion of this in the 0.9 degree stepper thread: https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../stepper-motor-upgrades-to-eliminate-vfa-s-vertical-fine-artifacts/

leptun, 2019-10-31:
Have you tried setting the microstepping manually in the configuration and using the printer without the M350 mod? I want to make sure it's not me causing the issue :)
It is probably an overflow in the planner calculation for the steps required for homing. Movement becomes erratic at high microstepping values anyways in the firmware so this situation can/should be avoided.

leptun, 2019-10-31:
Looking at the thread, they seem to have problems with stallGuard and the Moons' motors. I also have some problems with my modded TMC2130 miniRambo 1.3a board, but I wasn't ever able to make it reset.

vintagepc, 2019-10-31:
Yes, I'd get this before the M350 feature existed as well. 

My setup has X/Y microstepping at 8 and the steps/mm still at 100. I don' think that'd overflow anything since it's the same net step calculations. I know that the einsy definitely can't handle 16 usteps and 200 steps/mm at fast travel. 

There's a wealth of info on this in the linked thread in regards to stallguard tuning, and some here as well:
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../tuning-homing-for-0-9-degree-motors/

vintagepc, 2019-10-31:
Just saw your update. I have OMC/Stepperonline motors. They work quite nicely with stock firmware once you have the ability to set usteps and figure out that belt tensions need adjusting if homing fails.

I suppose I could have filed a bug earlier when I first encountered this, but it never occurred to me since I was already off the beaten path and I try to keep my support expectations for unofficial alterations minimal :)

vintagepc, 2019-10-31:
For reference, here's a normal home serial output:

> SENDING:G28 Y
tmc2130_home_enter(axes_mask=0x02)
echo:busy: processing
echo:busy: processing
  0 step=51 mscnt= 826
tmc2130_goto_step 1 35 2 1000
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02

and here's a failed home where the printer resets:

> G28 X
SENDING:G28 X
tmc2130_home_enter(axes_mask=0x01)
  0 step=12 mscnt= 193
tmc2130_goto_step 0 43 2 1000
start
echo: 3.8.1-RC1-2851
echo: Last Updated: Oct  5 2019 12:35:45 | Author: (none, default config)
Compiled: Oct  5 2019
echo: Free Memory: 2112  PlannerBufferBytes: 1392
echo:Stored settings retrieved
adc_init
CrashDetect DISABLED
FSensor ENABLED
Sending 0xFF
echo:SD card ok
MMU => 'start'
MMU <= 'S1'
MMU => '106ok'
MMU <= 'S2'
MMU => '372ok'
MMU version valid
MMU - ENABLED

Video of the failure:

https://vimeo.com/370181588

vintagepc, 2019-10-31:
Peeking at the code, looks like it's dying in this region: (2220 of marlin_main.cpp)
tmc2130_goto_step and home_exit print to serial; the latter never happens during failure.

		uint8_t back = tmc2130_home_bsteps[axis];
		if (tmc2130_home_enabled && (orig <= 63))
		{
			tmc2130_goto_step(axis, orig, 2, 1000, tmc2130_get_res(axis));
			if (back > 0)
				tmc2130_do_steps(axis, back, -axis_home_dir, 1000);
		}
		else
			tmc2130_do_steps(axis, 8, -axis_home_dir, 1000);
		tmc2130_home_exit();

if your guess as to an over or underflow is correct then the suspect is probably this while loop from the goto_step routine:

	while ((cnt--) && ((mscnt >> shift) != step))
	{
		tmc2130_do_step(axis);
		delayMicroseconds(delay_us);
		mscnt = tmc2130_rd_MSCNT(axis);
	}

vintagepc, 2019-10-31:
... We have a winner.

 I made this change to the above while:

while ((cnt--) && ((mscnt >> shift) != step))
	{
		printf_P(PSTR("tmc2130_goto_step_while %d %d %d %d \n"),cnt, mscnt ,shift, step);
		tmc2130_do_step(axis);
		delayMicroseconds(delay_us);
		mscnt = tmc2130_rd_MSCNT(axis);
	}

[capture.txt](https://github.com/prusa3d/Prusa-Firmware/files/3795867/capture.txt)

vintagepc, 2019-10-31:
Oops, updated capture correctly printing the values as unsigned ints:

[capture2.txt](https://github.com/prusa3d/Prusa-Firmware/files/3795942/capture2.txt)

vintagepc, 2019-11-01:
I spent some more time noodling over this and unfortunately I'm not familiar enough with expected/desired behaviour and TMC programming to know what is necessary to fix this situation; my gut is that the while expects cnt will always start >0 but we've found a corner case where it is exactly 0 - I'm just not certain whether fixing the while to check <=0 is just putting a bandage on the symptom. If you have insights to point me in the right direction I'd be happy to take this on with some guidance.

leptun, 2019-11-01:
I will take a look into the issue this weekend.

DRracer, 2019-11-08:
@leptun if you find a solution, we can still merge it into 3.9

vintagepc, 2019-11-22:
I've pushed up a candidate fix in #2342 .
This particular corner case seems to be most aggravated when the axis doesn't get a good solid hit, e.g. there is a zip tie or something else in the way. 

I suspect this fix may have minor impact on homing repeat-ability, but given that the primary cause of something squishy in the axis path already, I wouldn't be too worried about it in the first place... :-/
Alternately, we might be able to use it to issue a warning to the user ("Homing failed, ensure axis is ....") but that would require additional research to understand and I don't think my sample size of one printer is sufficient.

commit d6ac9e46 (2019-12-15): Merge pull request #2342 from vintagepc/2306-hard-reset-homing-fails

Fix #2306 hard reset homing fails
closed 2019-12-15 by DRracer