prusa3d/Prusa-Firmware #2310: Changing filamment after runout during print - adjust speed of extruder

ivantichy, 2019-11-02:
When filament runs-out during print user follow procedure to continue with new filament. Happy scenario is OK but when you select that new filament was not loaded correctly extruder tries to load more filament. The problem is that this step contains two parts - firstly extruder goes pretty fast (assuming it is loading filament into empty hotend) and then it slows. This is OK for loading into empty hotend but when tried in situation where filament is already loaded this cause extruder to skip.

MK3 prusa3d_fw_MK3_3_8_0_2684.hex

vintagepc, 2019-11-02:
While not experiencing this directly, I've noticed some filament change menus have 3 options, the 3rd being "it's loaded but the colour isn't right yet" which has the effect of purging more (but not at a high speed). I have also encountered menus where this option is not present and the answer is only yes/no. 

If your issue is on this "2-choice" change menu, I wonder if unifying them to all have the third choice would help fix this issue.
If you have the choice to pick the "colour not correct" option, does it also cause this skipping for you?

ivantichy, 2019-11-02:
There are two options: 1) yes and 2) It is it's loaded but the colour isn't right yet - and this loads at high speed.

closed 2023-10-10 by github-actions[bot]