prusa3d/Prusa-Firmware #2311: Minor Power Panic Issue (Mk3s -FW 3.8.1)

Tanischacat2009, 2019-11-02:
After a very brief power interruption yesterday morning (sag), which was just long enough to cause everything in the house to shut off and on (computers, clocks etc.), the printer did the same, but came back on right away.  Everything worked correctly and the printer resumed a very long print in which I was working on, but the power interruption caused a 1 mm layer shift from left to right on my project, which renders it imperfect versus ruined.  Nothing is perfect and I don't expect it to be, but just wanted to point out that this could be an issue for some people.

leptun, 2019-11-02:
Did you run selftest on the printer? It also calibrates the Homing after it checks everything. Which axis shifted? If it is X, then check if you can adjust the bundle of wires, maybe it is hitting something and causes a softer home, making it less reliable.

Tanischacat2009, 2019-11-02:
I ran the self test and full gamut of calibrations on the printer when I updated the firmware last weekend.  All calibrations passed, including the self test.  I have had many successful prints in recent times and at close to maximum height too for the printer.  I'll investigate the wire issue in which you had mentioned.  The printer worked perfect before and after the power interruption, it's just the small shift has left a noticeable flaw on my project.  Yes, it was an x-axis shift.  I had also adjusted the belts on the printer recently too and there was no blobs which would also explain a shift.

moojuiceuk, 2019-11-03:
I've had a similar same thing happen to me on FW 3.8.1 RC1 - 2851. Power dropped out for the first time whilst printing and the RCD for the circuit tripped (often the RCD trips if the mains externally drops and the lights flicker). Restored power, printer re-heated and continued off. I managed to wipe the nozzle just before it touched the work too, but looking closely there is a very small but noticable line in the print. It feels like over extrusion rather than actual layer shift as the bead runs all round the perimeter of that layer with a noticable ridge / bump. I can only assume the same line was re-printed again as the power must have failed just before the layer change?

Tanischacat2009, 2019-11-03:
There seems to be some noticeable difference in the print, even if it's paused and resumed again.  The power outage happened overnight so I wasn't able to intervene in any way when it happened and one can't be babysitting a printer 24/7, especially when it is to run for 22 days straight, like this project is.  I am still printing the same project, imperfect as it is now because of my reluctance to scrap it and restart after 5 and a half days of printing...and then to gamble on the power not even being interrupted even once over 22 days too is a very tall order.  No power system is perfect 100% of the time, and nor are 3d printers and firmware that gets flashed into them, so that is why I am considering an uninterruptible power supply unit to safeguard against power issues, even for the shortest and most common outages.

DRracer, 2019-11-08:
@Tanischacat2009 we have seen x-y layer shifts after power panic during the tests, but only as @leptun writes - when the machine wasn't xyz calibrated correctly.
Anyway, thanks for reporting, we'll try to reproduce your issue on the FW 3.8.1

Tanischacat2009, 2019-11-08:
My printer is in an enclosure, so it is possible that the wires on the left hand side could have been too close to the wall and caused a small shift during carriage movement.  This is the first and only print that I have done since installing the latest firmware and calibrating the machine.  After more than 12 days of total print time, it is still running flawlessly since the power outage mentioned in this thread had happened about a week ago.  I am ordering an uninterruptible power supply unit, so if there is a layer shift in any of my future prints, I would attribute it to belts, a calibration issue or perhaps wires.

closed 2019-11-11 by Tanischacat2009