prusa3d/Prusa-Firmware #2313: MMU2 retracts before extruder unloads in filament runout (Mk3S, 3.8.1/1.0.6)

vintagepc, 2019-11-02:
Just had this occur when my MMU print ran out of a certain colour mid print. Runout detection worked as expected, but the MMU did its retract cycle *before* the extruder had completed its unload. As a result there was no filament in the MMU for it to grab and retract. 

 I had to remove the bowden connector from the selector and pull the filament out manually before I could load replacement filament.

Please let me know if I should move or crosspost this to MM-control-01; not entirely sure whose domain this falls under but my suspicion is that the Mk3S firmware is the one telling the MMU to retract.

gudnimg, 2023-07-04:
@vintagepc do you think we can close this Github issue when 3.13.0 is released? Or is it still a relevant issue today.

Prusa-Support, 2023-09-20:
@vintagepc @gudnimg 
This feels obsolete at this point so I'll close it but it can definitely be opened back if needed. 

Michele Moramarco
Prusa Research

closed 2023-09-20 by Prusa-Support