prusa3d/Prusa-Firmware #2314: After Filament runs out... it extrudes the new filament into the printing part (Mk3S 3.8.1/MMU2S 1.0.6)

RolfHodl, 2019-11-02:
Just run out ot filament during a long print... the mmu2s just extracted the left over filament right out. But the loading was horrible. Refill the filament and the mmu2s made a filament load - but then no more. Had to manualy load from the mmu2s to the priting head. After a few tries it starts again - but now it just prints into the actual print and then ask - "is the filament in the right colour". 

So here are 2 issues:
-the mmu2s don't load the filament to the printing head just relaods it
-after reload it just prints a big blob on the resuming point - this should be done far away and then start reprint

same over here:
https://github.com/prusa3d/Prusa-Firmware/issues/2297
and this is the other problem
https://github.com/prusa3d/Prusa-Firmware/issues/1571

vintagepc, 2019-11-03:
Interesting, I had the exact opposite happen in #2313... it moved to the front right as expected and loading went fine. In my case the unloading was what failed. 

Sounds like an underlying timing issue could explain both situations...

huskyte, 2019-11-08:
In my case it always blobs into the weep tower which rips it off the bed most of the time. 
Here is what I posted in the Prusa forum (https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-multi-material-hardware-firmware-and-software-help/blobbing-on-weep-tower-after-filament-run-out-detected/#post-172819):

"...
I will update this topic as soon as I know what's needed. Anyway, the PRUSA with MMU2s is running great, but occasionally has a filament run out detected. Im running the latest firmware here. 

It asks for the unload and reload procedure, squeezes a bit of filament out (parking position front right), then moves back to the weep tower. While moving I can already see some additional filament being squeezed out (if you are quick it can be removed on the fly). It is clearly no oozing. On the weep tower, while the head doesn't move but is already lowered to its surface, another big material blob is generated. The weeping starts and usually ends up at best in a recoverable crash detection. Quite often the weep tower becomes loose.

My first theory was: The fake filament run out was detected by a piece of broken off filament so that the retracted length to the FINDA did not match, hence run out detected (seems to happen during load/unload). The tip would then be in the extruder being pushed by the new filament. BUT, this cannot be true, not if you do another run aka filament color not correct. The additional volume of filament should at some point squished out and the (hopefully) retracted filament length match the normal length. 

There simply seems to be no way to actually ensure no blob being generated on the weep tower. 

So the question is: Is that a common problem?

And the second question: How do I suggest to make a change in the firmware?"

This should be handled differently. No idea by what the extrusion command is caused, but it seems more than useless while just sitting on the weep tower not moving sideways at all.

RolfHodl, 2019-12-09:
It was a single color print with mmu2s just to clarify.

closed 2023-10-10 by github-actions[bot]