prusa3d/Prusa-Firmware #2316: Is the powerpanic feature supposed to work when printing through Octoprint?

mmoraes80, 2019-11-08:
I was running a print in normal mode through Octoprint the other day and the power went down for like 10 seconds but the printer did not lift the nozzle and when it turned back on, it did not asked to continue the previous print job and simply showed the main info screen on the LCD display. 

Since my Octoprint server is not using the printer SD card, I believe it was the reason why the job wasn't saved.

That being said, is it possible to integrate the feature regardless the print job is from the SD card or Octoprint?

leptun, 2019-11-08:
No. Unfortunatelly it is impossible since octoprint also looses power and it doesn’t save the last position like the printer when printing from sd.

cryocide, 2019-11-10:
Yeah, the issue there is going to be that even if Octoprint kept a rolling log of the last few lines of gcode sent to the printer just in case power was lost, it would still be difficult, or maybe even impossible, to know which line was the last successfully completed command because of command buffering via USB/Serial.  There would have to be some strong two-way communication going to support such a thing.  Octoprint can monitor SD print jobs, but loses quite a lot of functionality for plugins such as OctoLapse when doing so.

amoose136, 2019-12-01:
What if on power loss the printer stores all lines in queue to a file on the SD card if one is plugged in? Then when octoprint reboots it could check if a stub file exists and if it does, octoprint can figure out what to do from there. The only thing remaining would be a few comment lines thrown into the stream periodically so octoprint could know what lines it was feeding without having to constantly do a creality-style sd backup on its side as well.

closed 2023-10-10 by github-actions[bot]