prusa3d/Prusa-Firmware #2324: Mesh bed leveling correction

mmoraes80, 2019-11-12:
Hi,

The current mesh bed leveling correction is done as front, back, right and left. Why not make it as quadrants instead? This way, if one wants to correct the front for instance, simply adjust the 2 quadrants in the front. However, if the adjustment is required only at one of the corners, the respective quadrant could be adjusted without ruining the other opposite side.

[image]

william14235, 2019-11-23:
I think it's because it is more accurate with nine sections, and 3d printing needs extreme and precise accuracy.

mmoraes80, 2019-11-25:
Where are you getting nine sections? The accuracy should come from the PINDA sensor, but it does not necessarily flattens the virtual bed 100%, so the extra adjustment by corner would be helpful.

HarrySatt, 2020-02-11:
I would like to see the algorithm which takes account of the values set with the mesh bed correction values. what I see so far is that the middle position of the mesh bed is not hold fixed to the z level correction. Is this right?

closed 2023-10-10 by github-actions[bot]