prusa3d/Prusa-Firmware #2325: Power Panic not working when print is paused

JKinch03, 2019-11-14:
In the latest version when I pause a print and I unplug the printer from the wall, the power panic doesn't kick in. It wouldn't usually need to be in use when the printer is paused, but what would happen if you were to trip over the power cord or experience a power outage after you paused it. 

A simple fix *(from my point of view)* to this would be to save the current positions when the user tells the printer to pause. Then, if it were to become unplugged while paused, it would take that position and start from there when it turns back on. Once that all happens, the printer would continue as if a normal power panic just occurred.

william14235, 2019-11-23:
Actually, I'm pretty sure power panic only works when it is not paused. I tried power panic before, but I paused the print, so it did not save correctly. I think that if you need to move and unplug the printer, you should just unplug it and let it power panic. However, I do agree that Prusa should let it pause and resume even after it power panics. But maybe from their point of view, if it power panics, it will move just slightly and layer shift, so if they fixed it they would also have to make it calibrate every time after it power panics or pauses.

wavexx, 2020-01-27:
This is a know bug. PowerPanic will currently not resume correctly during:

- a paused print (also  #2249)
- during the middle of a filament change (#2001) 
- mesh bed leveling (just tested: purge line prints on the wrong spot)