prusa3d/Prusa-Firmware #2326: [3.8.1] [MK3S] [MMU2S] Debug message during filament change: "OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded."

cryocide, 2019-11-15:
I think my printer may be haunted.  I'm getting some interesting, but not exactly useful debug messages in my terminal window via USB on 3.8.1 FW on my MK3S MMU2S during filament changes. Busy/wait messages are suppressed for the sake of keeping this legible.

> `Recv: ok
> Send: N25052 T1*59
> Recv: mmu_get_response - begin move: T-code
> Recv: MMU <= 'T1'
> Recv: Unloading finished 2
> [...]
> Recv: mmu_get_response - begin move: load
> [...]
> Recv: MMU <= 'A'
> Recv: MMU => 'ok'
> Recv: mmu_get_response() returning: 1
> [...]
> Recv: MMU can_load:
**> Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.**
> Recv: ok
> Send: N25053 M105*38
> Recv: ok T:230.1 /230.0 B:90.2 /90.0 T0:230.1 /230.0 @:63 B@:46 P:56.4 
> 
> `

I think this output is coming from code in the mmu.cpp file of the source code, but I'm not sure what's causing the variable to output this message.

`DEBUG_PUTS_P(PSTR(" succeeded."));`

leptun, 2019-11-15:
That is the MMU load debugging. If it shows OOOOOO... than that is right. However if it shows OOoooOooooOoooOOOO then your IR sensor is not detecting the filament propperly and a load failure is issued.
O is filament at gears.
o is no filament at gears.

succeded. is after the T command is finnished successfully.

cryocide, 2019-11-15:
I see, so not a bug in debugging! I can see how this would be useful if the IR sensor did not stay triggered.  In that case, there is no need to keep this as an open issue, but maybe good to have just in case someone else like me searches for it.  Thank you!

closed 2019-11-15 by cryocide