prusa3d/Prusa-Firmware #2328: Minor: Wrong "File incomplete" reporting

Panayiotis-git, 2019-11-15:
The incomplete file check, searches for the M84 command at the last 20000 bytes of the gcode file.
Since the Slicer generated files contains a ton of comments with the used configuration at the end of the file, there are cases when the 20000 bytes slot misses the M84 command.

Check the attached sample. I have defined 15 extruders.

Possible solution could be to skip the comments.

[FileIncomplete.zip](https://github.com/prusa3d/Prusa-Firmware/files/3852914/FileIncomplete.zip)

michalxfanta, 2019-11-25:
Thanks for reporting this issue.

Panayiotis-git, 2023-10-05:
I still get the false warning, but I can continue with the print

gudnimg, 2023-11-19:
Assigning this to myself to the issue doesnt get closed. 

@Panayiotis-git have you looked into implementing a fix for this? I wonder why the firmware can’t parse the entire file line-by-line.

gudnimg, 2023-11-19:
I’ve seen on Prusa Forum and Facebook reports thst other people are still seeing this.

3d-gussner, 2023-11-20:
https://github.com/prusa3d/Prusa-Firmware/issues/2328 solves the issue.

closed 2023-11-20 by 3d-gussner