prusa3d/Prusa-Firmware #2330: Cannot complete calibration after upgrade on mk1

Bwsusa, 2019-11-16:
continually get calibration message when powering on printer since update. none of the calibration steps will resolve issue Printer is Genuine Prusa i3 mk1.

3d-gussner, 2019-11-17:
@Bwsusa MK1?

Bwsusa, 2019-11-18:
Yep. Mk1 with the orange bed leveling knobs.

Will W.

william14235, 2019-11-23:
yes, my bed was like that too, even though it is an mk3. for some reason, the right side of the bed was higher than the left, unless it was just an error. I simply raised the live adjust z, but if you can level the bed, try doing that first.

Mafaldone, 2020-05-10:
yeap. Me too. I have the same issue after upgrating to new firmeware, my mk1 continue to display that is not calibrated. Moreover, this break also the connection to my octoprint. Timeout on serial connection.
Please provide us a work around!

Mafaldone, 2020-05-10:
> yes, my bed was like that too, even though it is an mk3. for some reason, the right side of the bed was higher than the left, unless it was just an error. I simply raised the live adjust z, but if you can level the bed, try doing that first.

I think that this is not related with bed leveling. It is related to the fact that printer starts always as not cofigured properly, preventing serial communication over usb to happen.

3d-gussner, 2020-05-11:
@Mafaldone Do you have a MK1? If so please use that latest stable firmware for the MK1.

Mafaldone, 2020-05-11:
@3d-gussner yes I do own an mk1. I used the 3.0.12 version as you can see in the driver and firmeware page https://www.prusa3d.com/drivers/ 
. Is the 3.0.12 considered stable?

Mafaldone, 2020-05-12:
Hi all, I just solved it. After upgrade you should print the V2calibration.gcode file. Within that file there are two magic gcode G87  that set the value to live z axes. More info in https://mattshub.com/blog/2017/06/04/v2calibration-bypass 
After printing V2calibration.gcode and restart the octoprint server, I am able to connect without timeout. Hope this help others as well

closed 2023-10-04 by github-actions[bot]