prusa3d/Prusa-Firmware #2333: [MK3s][3.8.1][1.0.6] Random Resets and Firmware Crashes

sarf2k4, 2019-11-16:
I have this printer for over a year now, Using MK3s v3.8.1, and mmu2s v1.0.6.

This is not the first time I encountered random resets and firmware crashes. Often times they are triggered by the mmu2 having errors like loading filament to the nozzle, it could crash and reset itself.

The recent random firmware crash and rest, or should I say weird behavior, was last night when I had a job around 50% or so that has 3 hours left, the print itself was 7 hours total.

What happened last night was that the filament snapped inside the teflon tube while the printer still continues to print without knowing there aren't any filament going through the bondtech gears despite having sensors there. I initiated "change filament", it pulled the filament out of the mmu2 and I trimmed it. trying to load it back, by pressing the knob, it tries to load the filament into the nozzle, suddenly the lcd screen goes blank. The hotend fan still running, I waited for about 1 minute and I even pressed the left and right mmu2 button, the selector moves like it should be when not printing. I just decided to press the reset button on the lcd panel and there I go, the print itself failed because of this weird behavior of "change filament". Only 2 pictures and I don't think I can give more proof than this anyway.
[image]
[image]

Another some of the random behavior is, I think my finda/pinda probe for mmu2 starting to degrade where the printer detects filament runout. I solved the issues with a it of extra steps because the blob were deformed because I think the extruder motor were being hot. It did print okay, this one same thing to be printed, but successfully printed. But the weird behavior was, after completed print, I turn off the wall plug switch, the printer made the "power panic" beep. This is similar to the issues in firmware v 3.7.x or 3.8.0 I think where the beep will persist even if the printer is not printing.

Please check this issues and I am not fond of this thing. I believe it may be due to insufficient power delivered by the power supply unit to power up both the main printer electronic along with mmu2. Further the extruder motor being quite that hot, it must be pulling quite a lot of amps out of the whole thing probably.

william14235, 2019-11-23:
That has happened to me too. The problem for me with the random resets was not the firmware, but the hotend heater cartridges' bare wires touched and short circuited. If this has happened to you, you should separate the wires.

wavexx, 2020-01-27:
I don't have an MMU, but there were multiple issues related to pausing the printer (which is also happening behind the scenes while doing a filament change) which should have been fixed with #2274

closed 2023-10-04 by github-actions[bot]