prusa3d/Prusa-Firmware #2337: Can't Compile - help!

AbeFM, 2019-11-18:
I've been trying to help another group with a fork of this project, and while trying to make a small tweak I couldn't get the firmware to compile. After a lot of digging, I've downloaded the latest release code zip file... and I get the same error when trying to compile.

You can check out my issues at https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/issues/167 and for anyone who's interested, ZeroBeast's firmware is worth a look.

Anyway, the specific issues I'm having:

I cloned the latest repository, and selected the RAMBO board, so I would expect this to compile. When I try, however, I get:
``In file included from sketch\Configuration.h:62:0,

                 from sketch\Configuration.cpp:1:

eeprom.h:228:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression

 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500``

i've gone through and added bits to platform.txt and other voodoo.... I'm not sure why this won't compile. An older version (370 from Prusa) with some tweaks I did compiles fine. 

Here's the old one that compiles:
while there are a lot of changes, the one that leaps out at me is:
static M500_conf 
becomes
static constexpr M500_conf 

`````
#define EEPROM_FIRMWARE_PRUSA_MAGIC 0

#ifdef __cplusplus
#include "ConfigurationStore.h"
static M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
#endif

#endif // EEPROM_H
`````

And the new one that doesn't:

`````
#define EEPROM_FIRMWARE_PRUSA_MAGIC 0

#ifdef __cplusplus
#include "ConfigurationStore.h"
static_assert(EEPROM_FIRMWARE_VERSION_END < 20, "Firmware version EEPROM address conflicts with EEPROM_M500_base");
static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
static_assert(((sizeof(M500_conf) + 20) < EEPROM_LAST_ITEM), "M500_conf address space conflicts with previous items.");
#endif

enum
{
    EEPROM_MMU_CUTTER_ENABLED_enabled = 1,
    EEPROM_MMU_CUTTER_ENABLED_always = 2,
};

#ifdef __cplusplus
void eeprom_init();
bool eeprom_is_sheet_initialized(uint8_t sheet_num);
struct SheetName
{
    char c[sizeof(Sheet::name) + 1];
};
void eeprom_default_sheet_name(uint8_t index, SheetName &sheetName);
int8_t eeprom_next_initialized_sheet(int8_t sheet);
void eeprom_switch_to_next_sheet();
#endif

#endif // EEPROM_H
`````

Even if I move the preceding line
static_assert(EEPROM_FIRMWARE_VERSION_END < 20, "Firmware version EEPROM address conflicts with EEPROM_M500_base");
to after the statement that throws an error, I still can't get it to compile.

If I remove the "constexpr", it compiles, but I get a bunch of subsequent warnings of the form

```
sketch\mmu.cpp:113:13: warning: 'mmu_idl_sens' was declared 'extern' and later 'static' [-fpermissive]

 static bool mmu_idl_sens = false;

             ^~~~~~~~~~~~

In file included from sketch\uart2.h:14:0,

                 from sketch\mmu.cpp:6:

sketch\mmu.h:8:13: note: previous declaration of 'mmu_idl_sens'

 extern bool mmu_idl_sens;

             ^~~~~~~~~~~~

sketch\mmu.cpp: In function 'void mmu_continue_loading()':

sketch\mmu.cpp:1112:13: warning: 'void mmu_continue_loading()' was declared 'extern' and later 'static' [-fpermissive]

 static void mmu_continue_loading(void)

             ^~~~~~~~~~~~~~~~~~~~

In file included from sketch\uart2.h:14:0,

                 from sketch\mmu.cpp:6:

sketch\mmu.h:92:13: note: previous declaration of 'void mmu_continue_loading()'

 extern void mmu_continue_loading(void);

             ^~~~~~~~~~~~~~~~~~~~

sketch\ultralcd.cpp:114:13: warning: 'void lcd_language_menu()' used but never defined

 static void lcd_language_menu();

             ^~~~~~~~~~~~~~~~~

Sketch uses 220122 bytes (85%) of program storage space. Maximum is 258048 bytes.
Global variables use 6517 bytes of dynamic memory.

```

It's quite possible I'm doing something dumb... but I don't know what - and I would really appreciate some pointers on how to get the firmware compiled.
Windows10, ArduinoIDE, MK3s/MMU2s - everything current.

3d-gussner, 2019-11-19:
@AbeFM As far i am aware TheZeroBeast firmware is compiled using Platform IO.
To compile Prusa Firmware please read the README . I think easiest is to use `PF-build.sh`script we wrote to get a known/stable build environment for the Prusa Firmware. If you want to do it manully:

- Use Arduino IDE 1.8.5 and NEVER EVER update the "AVR Boards" in Boards Manager. Best disable in properties the Check for updates.
- Use Arduino IDE in portable mode. By default Arduino IDE uses shared folders, which may compromise your Prusa Firmware environment.
- Follow the instruction to add the Prusa json files url.

For the MMU2s follow the https://github.com/prusa3d/MM-control-01/blob/master/README.md instructions.

AbeFM, 2019-11-19:
I was thinking since 3.7.0 compiled that the environment was ok.

I'd mentioned this to TZB, he cleaned stuff up a bit from his PlatformIO version and now I only get one warning:

```

sketch\ultralcd.cpp:114:13: warning: 'void lcd_language_menu()' used but never defined

 static void lcd_language_menu();

             ^~~~~~~~~~~~~~~~~

```

It seems a bit more tenable. Probably that could have some "ifdef" statements to clean it up. I posted here because I'd seen similar issues with the PR version.

It sounds like people aren't having issues with it compiling in Arduino IDE? In the new TZB which DOES compile, he pulled out he same "constexpr", I'm not sure what he did to rid himself of the other errors - but again, I saw the same errors above when compiling the official PR version.

Parker-Drouillard, 2020-02-02:
I'm also having the same issue;
eeprom.h:236:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression

I am working on my own fork of the latest MK3 software for our industrial dual extruder versions of the mk3. Currently the machines are running off the old mk2.5 fork which is about 3300 commits behind the latest chain (I didn't realize the way it was laid out on their github and grabbed the wrong by mistake)

I've just spent the past couple days cleaning up their code to make it more usable, and then upon first compile ran into this issue.

Upon further inspection, even with a fresh download & following all their instructions, I am also encountering the issue as mentioned per the above. Guess something is wrong with my environment now.. 

Not sure if anyone found an actual cause or solution to this yet?

riskable, 2020-06-23:
I've been compiling the firmware successfully for ages but this one just cropped up for me too and I can't seem to fix it.  I did a fresh checkout and used all default settings for everything and I get that same error:

```
riskable@somehost:/storage/Prusa-Firmware $ ./build.sh 
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
In file included from sketch/Marlin.h:45:0,
                 from sketch/BlinkM.cpp:5:
sketch/MarlinSerial.h:85:29: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
 #if UART_PRESENT(SERIAL_PORT)
                             ^
sketch/MarlinSerial.h:85:29: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
sketch/MarlinSerial.h:85:29: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
sketch/MarlinSerial.h:85:29: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
sketch/MarlinSerial.h:85:29: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/BlinkM.cpp:5:
eeprom.h:584: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
```
I'm just doing the regular Linux build (using build.sh).  Not even using the Arduino software (as far as I know).

leptun, 2020-06-23:
You are using a version of arduino that is too new. It must have gcc version 4.x.x, not 5.4.0 as it is with the latest arduino.
In any case, I actually have a branch with fixes for the new gcc, but I don't think it's completely stable. Had it crash two times during printing for no obvious reason.

Parker-Drouillard, 2020-06-24:
Exact what Alex said.

I will be at work soon and can let you know the exact Arduino version which
worked for me if you’d like!

-Parker

-- 
*________________________________________________________________*
Thanks Again,
Parker Drouillard

Owner/Founder
*________________________________________________________________*

*________________________________________________________________*

PepCorp.
Additive Manufacturing Solutions Division
401 Sunset Ave, Windsor Ontario
CEI Building, RM 1133 (First Floor)
N9B 3P4

Tel     - 519 300 2796
Web  -  p <https://www.3dhubs.com/service/PepCo>epcorp.ca
Email - Info@PepCorp.ca
*________________________________________________________________*

3Dietrich, 2020-07-24:
I'm either too stupid or there is just no way on mac 10.15..?
I've gone a round of up- and down-dating and each time there was a limit. It seems that for macOS 10.15 Catalina there is no way to compile Prusa firmware...?!
 - 1.8.5 (demanded by Prusa) - too old, it compiles nothing (Arduino problem on MacOS 10.14++)
 - [special version](http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-8976-BUILD-869-macosx.zip) (actually 1.8.10) - doesn't compile, too new for Prusa
 - 1.8.11 (current release) - too new for Prusa

?

leptun, 2020-07-24:
At the moment the official firmware can’t be compiled on the latest macOS versions. You ether gave to compile it in an older macOS version, on linux, on Windows or you may try to use this branch with the latest arduino in your branch: https://github.com/leptun/Prusa-Firmware/tree/MK3_GCC

3Dietrich, 2020-07-24:
Thank you Leptun! 
What does _GCC mean, what is different? Why is it in your area, not Prusa?

leptun, 2020-07-25:
It’s in my area because it’s not official. I named it “GCC” as it includes fixes in code that make it compatible with newer gcc and g++ versions, mainly the one used in the newest arduino ide. As a result, you should be able to compile this one on the mac.

3Dietrich, 2020-07-25:
thank you Alex,
FYI: there is also the same hint _It is recommended to use version "1.8.5"_ blabla

I tried it but there is another problem (I added bugreport there)

leptun, 2020-07-26:
Oh... I didn't update the documentation there at all. Just the firmware bugs are fixed. You still have to use the latest arduino regardless of what version is recommended there. The rest instructions still apply with minor changes if using the scripts.

davidewen, 2020-11-06:
I'm having the same trouble that everyone else has reported with an extra.
I'm on MacOS 10.15.7
I've tried verifying leptun's MK3_GCC - this is the one with the following error. I at least made it past the eeprom error.

exit status 1
'reinterpret_cast<M500_conf*>(20)' is not a constant expression

Parker-Drouillard, 2020-11-06:
Try rolling back your arduino IDE version. I can check what version I'm
running if you would like. In the end I think I ended up having to change
some of the core library files to get it to work but it's been so long I
can't really be of much help towards what/why

davidewen, 2020-11-06:
> Try rolling back your arduino IDE version. I can check what version I'm running if you would like. In the end I think I ended up having to change some of the core library files to get it to work but it's been so long I can't really be of much help towards what/why
> […](#)

Thanks Parker.

Rolling back to 1.8.10 did the job verifying FW version MK3_GCC

MikePfaff, 2020-11-30:
Ran into the same problem on MacOS 10.15 and Arduino 1.8.12 and fixed it by changing line 594 in the eeprom.h from
`static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500`
to
`static M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500`

(This is the fix made by the MK3_GCC fork, see also in diff https://github.com/prusa3d/Prusa-Firmware/compare/MK3...leptun:MK3_GCC#diff-e09465d9901d088a6392810e5cfbda16fee36d4adbc8e34936117c78a32cdcb4 )

lensdigital, 2021-01-21:
> Ran into the same problem on MacOS 10.15 and Arduino 1.8.12 and fixed it by changing line 594 in the eeprom.h from
> `static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500`
> to
> `static M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500`
> 
> (This is the fix made by the MK3_GCC fork, see also in diff [MK3...leptun:MK3_GCCdiff-e09465d9901d088a6392810e5cfbda16fee36d4adbc8e34936117c78a32cdcb4](https://github.com/prusa3d/Prusa-Firmware/compare/MK3...leptun:MK3_GCCdiff-e09465d9901d088a6392810e5cfbda16fee36d4adbc8e34936117c78a32cdcb4) )

This solved issue for me on Windows 10 running Arduino 1.8.10! Thank you Mike!

3d-gussner, 2021-01-22:
@lensdigital As you are using Windows 10 please consider to use PF-build.sh with git-bash or WLS as described:
https://github.com/prusa3d/Prusa-Firmware#using-linux-subsystem-under-windows-10-64-bit
https://github.com/prusa3d/Prusa-Firmware#using-git-bash-under-windows-10-64-bit

The script will download and configure a stable and know build environment to build Prusa-Firmware.

closed 2023-10-12 by github-actions[bot]