prusa3d/Prusa-Firmware #2341: REQUEST/BUG?: missleading "MINTEMP fixed" message

Sedrah87, 2019-11-21:
Hi, 
my current Problem is: I have a broken bed thermistor wire.
In FW-version 3.8.0/3.8.1 (maybe other) I got midprint the message "MINTEMP fixed please restart" and the print stops. What I can't see there is a very short (1 sec?) message bevor "Err: MINTEMP **bed**"
So I replaced the extruder thermistor with no success. After several supportchats we could identify the problem by flashing 3.7.1 there came the message MINTEMP bed.
Maybe there is a way to show different messages for different broken thermistors. Think this will help all with a MINTEMP error.
Thanks

Kelly12612, 2019-11-21:
I've had this same problem with my new MK3S.  I added a comment on the assembly manual page, I do not think that this wire should be routed through the heatbed power wire clamp. Instead this wire should go outside of that clamp, and then be routed into the wire wrap from the side.
It's just way too hard to ensure that the wire stays in that tiny channel in the clamp while you're tightening the clamp, and the instructions/pictures suggest that you tighten that clamp completely until no gap is showing between the pieces.  This wire is far too fragile to withstand that.
Thanks,
Kelly

Tanischacat2009, 2019-11-22:
The thermister and heater cartridge wires attached to the hotend should have protection around them on the bottom closest to the hotend.  I have had six thermistor replacements since purchasing my mk3s last February and to prolong the life of the thermistor, I have resorted to using rubber heat shrink wire wrap around the thermistor and heat cartridge wires to add protection to them from rubbing, pulling or getting caught on things etc.  Though the wires themselves are relatively in expensive to replace, this is a somewhat substantial repair and more so if a hotend replacement is also required or easier to do at the same time.  Having the wires enclosed near the hotend would greatly reduce the amount of thermistor/heat cartridge wire replacements needed.

william14235, 2019-11-23:
I have not have any problems with my thermistor on my mk3, maybe there is a new design that has an effect on the extruder? I remember that the mk3 s is supposedly supposed to have a completely redesigned extruder.

nordurljosahvida, 2020-03-18:
The MINTEMP FIXED error should absolutely distinguish between hotend and bed by saying MINTEMP BED FIXED and possibly MINTEMP HOTEND FIXED

leptun, 2020-03-18:
The issue is fixed in the latest release. (3.9.0-RC1)

nordurljosahvida, 2020-03-18:
@leptun amazing thanks!

leptun, 2020-04-04:
@Sedrah87 Please close the issue.

Diedhert, 2020-04-18:
I have a new printer, two weeks old. I got this error, replaced thermistor, and still have the error "Mintemp fixed'....

3d-gussner, 2020-04-18:
@Diedhert which printer which firmware what is your ambient temperature. Can check under support for the temperatures? Or what temp does your lcd shows when it is not printing?

Diedhert, 2020-04-18:
HI @3d-gussner I have a Prusa MK3s which is about 3 weeks old now. Firmware is 3.8.1.
Ambient temperature must be 25-30° (4 printers in my room). 
I am not all the time in my room, so the last time it was cooling down and on 70°.

closed 2023-10-12 by github-actions[bot]