prusa3d/Prusa-Firmware #2345: Thermal runaway in MK3 (not S) when filament sensor is enabled.

alitibaliti, 2019-11-28:
Running (currently) the latest FW 3.8.1 on MK3.
When the filament sensor is enabled the printer randomly starts to lose temperature (usually both heaters) and reports thermal runaway. This happens on almost every print, usually in the first 5-6 minutes, sometimes during preheat.

The problem does not seem to be present when the filament sensor is disabled.

3d-gussner, 2019-12-02:
@alitibaliti Can you reproduce this? If so can you please write down your steps you have done? Does that also happens in older versions? Is it a stock printer and firmware?

leptun, 2019-12-02:
@alitibaliti are you printing from octoprint or from the sd card?

alitibaliti, 2019-12-03:
@leptun Printing from a sd card. No serial connection of any sort.

@3d-gussner Yes, it is easily (and unfortunately) reliably reproducible. 
It is a stock MK3 (w/o S upgrade) and stock downloaded FW 3.8.1.

Unfortunately I cannot recall the previous firmware version, but the printer was printing reliably. After not using it for over than 3 months I've upgraded the firmware. And the problems started.

**To reproduce:** Start any print with the filament sensor enabled. The problem usually manifests itself within the first 5-6 minutes of the print, occasionally later.

**What happens:** The temperatures on the display start dropping after while to a point there the printer stops with "Thermal runway" error message. 

With the sensor disabled I was able to run multiple long prints - some 11+ hrs. each.

My general feeling during 2 and a half days testing was that for some reason the heaters were shut down but the printer was printing until the temperature was low enough to throw a thermal runaway error. I haven't looked into the code, but assume there is different logic for the two kinds of filament sensors. And disabling it magically worked. It was more hunch than a logical test.

**Potential hardware issues:** I think I've eliminated the possible hardware reasons, please let me know if I've missed something:
- I've replaced the hotend heater;
- temporarily replaced the PSU and monitored the voltage out of it - stable 24 volts even when the temperatures were dropping down;
- a new thermistor is in the mail, but with the existing one I can see the temperatures slowly raising and dropping as opposite of suddenly dropping to extreme readings. And the values seem to match the actual temperatures. I can assume both thermistors are working.
- The filament sensor is not tested - I did not have a spare and have no way to check it's readings. (I assume it isn't like the new simple ON/OFF readings of the steel ball one).
The MK3S upgrade is on its way so I hope the new sensor or the MK3S version of the firmware to solve my problem.

The new version of the slicer was also a variable in my case, so I tried a Cura-sliced gcode with the same results.

Hope this helps. Please let me know if I can provide additional information.
Also if it is possible and useful and if you let me know how to do it I'll gladly capture any output over the serial connection. 
And thank you for your help!

alitibaliti, 2019-12-21:
I've upgraded my printer and no longer can reproduce the problem. Closing the issue.

closed 2019-12-21 by alitibaliti