prusa3d/Prusa-Firmware #2355: MINTEMP Error on hotend, when only the bed is being heated

noppej, 2019-12-04:
I am trying to heat my printer bed, while the hotend is still at room temperature (below 15 Celcius). The bed temp rises steadily, and then as it approaches the requested temp, the printer errors out with a MINTEMP error.

The attached Octoprint log shows the sequence of events ...
[octoprint (11).log](https://github.com/prusa3d/Prusa-Firmware/files/3922587/octoprint.11.log)

Prusa firmware 3.8.1, and Octoprint 1.3.12

michalxfanta, 2019-12-09:
This probably caused a low temperature in the room. The bed is cooling rapidly and that causing Min Tem error. The printer's working temperature range is 18°C - 38°C. Try to use an enclosure or raise the temperature in the room.

noppej, 2019-12-11:
Fully understood. Thanks.

closed 2019-12-11 by noppej