prusa3d/Prusa-Firmware #2356: Belt test feature is broken

vintagepc, 2019-12-04:
This feature does not work on my printer. If I try to use it, Z moves up ~20mm, and I get a "Selftest error" on screen instructing me to check X axis length. X never attempts to move, and the updated status value is 0.

Mk3S/MMU2 with 12/3 github pull & 1.0.6 firmware. 

Video: 

https://vimeo.com/377413263

3d-gussner, 2019-12-11:
@vintagepc
- Which printer? MK3s, MK3? I can see that you have a custom extruder.
- Which Firmware version?
- Stock or community or custom?
- Do you print from SD or via USB?
   - If USB how?
   - Did you try another SD card?
- What changed?
   - Did it work before (older firmware)?
   - Something else changed?
- Can you reproduce the issue? If so how?

vintagepc, 2019-12-11:
Errmm... did you mean to post that in reply to another issue? It's kinda non-sensical in this context.

> * Which printer? MK3s, MK3? I can see that you have a custom extruder.
> * Which Firmware version?
> * Stock or community or custom?

> Mk3S/MMU2 with 12/3 github pull & 1.0.6 firmware. 
(I have a few local tweaks that account for the E load distance difference. Nothing that should affect this)

> * Do you print from SD or via USB?
>   
>   * If USB how?
>   * Did you try another SD card?
Not sure what this has to do with a built-in firmware feature

> What changed?
> * Did it work before (older firmware)?
> * Something else changed?
Huh? It's a new feature, it did not exist in older firmware. First time it's here and first time trying to use it. 

> Can you reproduce the issue? If so how?
Yes. Exact same result every time I try to choose "Belt test" as shown in the video I get the shown result. (As noted, the standard "self test" still works fine)

vintagepc, 2020-01-17:
Tested in the 3.9.0 beta, the belt test feature still does not work for me.

Panayiotis-git, 2020-01-17:
I think that it also happened to me the first time I merged locally this PR. IIRC, a full printer reset was required to restore the functionality.

vintagepc, 2020-01-18:
Weird. I'm at a loss as to what would cause that though. I looked at the code and it just invokes the selftest subroutines - and the full self test does work fine.

ChipTechno, 2020-02-06:
I have the same problem.
It occurred event after i reset the printer by hold the knob on startup and M502.
I wonder if the belt is too tight that make the crash-detect triggered (even i disabled it on the setting)?

Panayiotis-git, 2020-02-08:
And this is the reason (https://github.com/prusa3d/Prusa-Firmware/issues/2472) why a full printer reset has fixed the problem for me :).

PR #2492 (2020-03-26): Belt test improvements (No display alteration)
Per @leptun's request, this PR contains only the the non-display-altering changes for the belt test fixes included in #2473 (i.e. fixes #2472 and fixes #2356 only)

No new translations were required for the display changes but we might want to do further cleanup of the display layout in this function before making that change. 

PFW-1102
PR #2473 (2020-04-20): #2472 #2356 #2484 belt test improvements
Forces high power mode for belt test.

- Fixes #2472, Fixes #2356, resolves #2484 

Edit: Also fixes #968 , closes #2547 

PFW-1104
closed 2020-03-26 by DRracer