prusa3d/Prusa-Firmware #2358: 3.8.1 Print head makes random moves

mdaneman, 2019-12-07:
After updating to firmware 3.8.1 the printhead takes random moves during print and then comes back. For example it may move all the way to the left, then move back and continue printing (leaving a   hole in the print since it was in the wrong place for a while).  This happens on nearly every print. Reverting to 3.8.0 fixes the issue.   I tried installing 3.8.1 and doing a factory reset on the printer - this still didn't fix the issue.
See the streak and gap in the middle square on the attached image

[image]

michalxfanta, 2019-12-09:
Could you send us a used G-code?

mdaneman, 2019-12-09:
Sure.  Here's the g-code.  However, I've had it happen with other prints as well.

[0p2_140mm_FiveSquare_0.4n_0.2mm_PLA_MK3S_29m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/3940540/0p2_140mm_FiveSquare_0.4n_0.2mm_PLA_MK3S_29m.gcode.zip)

3d-gussner, 2019-12-11:
@mdaneman
- Which printer? MK3s, MK3?
- Stock or community or custom firmware?
- Do you print from SD or via USB?
   - If USB how?
   - Did you try another SD card?

mdaneman, 2019-12-12:
Printer is MK3S using stock firmware. I printed via Octoprint. Didn’t check if printing from SD still has the issue or not.

JohnnyDeer, 2021-08-23:
@mdaneman, Is this issue still actual with latest FW version and PrusaSlicer?

closed 2023-10-10 by github-actions[bot]