prusa3d/Prusa-Firmware #2363: Printer restarts before print due to filament sensor panic

glowfishDE, 2019-12-09:
Hi,

after updates to the current firmware and octoprint 1.3.12 i cannot print as before anymore.

My printer runs a mesh bed leveling and does a purge line/bubble in the lower left corner before the prints usually.

Now it just reboots after mesh bed leveling and does not start the purge line.

I could find out, that this only happens when i have filament loaded into the MMU2S but NOT into the Nozzle.

Obviously contrary PINDA and FINDA readings confuse the current firmware and cause the printer to panic and reboot.

MMU2S says "i have filament" and printer says "no - i dont" and this is what crashes it.

When i "load (a filament) into nozzle" before starting a print, it works as expected. Even multimaterial prints run fine then. 

Imho it should load the filament into the nozzle by itself - it always has done that and loaded F1 (or what was selected in the slic3r STL file for the printed part.

I have all Filaments/extruders assigned correctly. The same GCODE file runs without crashing the printer when i manually load the filament into the nozzle before a print.

imho its a firmware bug because it always worked in the past. Printing since over a year like that. Now i upgraded and i does not work anymore.

vintagepc, 2019-12-12:
(This is a duplicate of #2301, I think.)

glowfishDE, 2019-12-12:
> (This is a duplicate of #2301, I think.)

Yes, that looks exactly like my issue. I´m closing this then.
Duplicate of 2301#

closed 2019-12-12 by glowfishDE