prusa3d/Prusa-Firmware #2364: Pronterface not working

Venthe, 2019-12-10:
I am no longer being able to connect to prusa via pronterface; I used to connect without a problem.
I can try to connect; **I get the "connected" response (prusa spins up)**; but no other command is acknowledged.

I've tried to connect via two PC's, both running 1909 ATM. Any ideas, guys? I have overextrusion problem which I cannot fix :)

Current firmware: 3.8.1-2869 official;
Software package 2.2.8

leptun, 2019-12-10:
I think that you’ve enabled the rPI port in settings. It must be disabled for the USB communication to be enabled.

Venthe, 2019-12-11:
Indeed it was! I don't recall ever fiddling around this setting, and I haven't changed it consciously for sure. Is it possible that it has been reset during flash? Or more likely, user (Me) error?

closed 2019-12-11 by Venthe