prusa3d/Prusa-Firmware #2365: Music tones are broken in version 3_8_1_2869

WhiteDragonD, 2019-12-11:
Copying from Prusa Slicer bug #3301, this might be a firmware issue.

To quickly test, just add the music code into a gcode file and 'print'.

I have now confirmed that version 3_7_2_2363 works but version 3_8_0_2684 on up does not.  Please fix.

Version
2.1.0

Operating system type + version
Windows 7/10

3D printer brand / version + firmware version (if known)
MK2.5s

Behavior
In Firmware version 3.5.1-1778, I could add the music tones at the end of a print (after M84 ; disable motors) and then it would play the music tones. With firmware version 3_8_1_2869, it just plays beeps without changing the tones.

Example of music tones inserted after M84:

;MarioOverworld
M300 S0 P75
M300 S2637 P150
M300 S2637 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S2093 P150
M300 S2637 P150
M300 S0 P150
M300 S3135 P150
M300 S0 P450
M300 S1567 P150
M300 S0 P450
M300 S2093 P150
M300 S0 P300
M300 S1567 P150
M300 S0 P300
M300 S1318 P150
M300 S0 P300
M300 S880 P150
M300 S0 P150
M300 S987 P150
M300 S0 P150
M300 S932 P150
M300 S880 P150
M300 S0 P150
M300 S1567 P150
M300 S2637 P150
M300 S3135 P150
M300 S1760 P150
M300 S0 P150
M300 S2793 P150
M300 S3135 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S2093 P150
M300 S2349 P150
M300 S987 P150
M300 S0 P600
M300 S3135 P150
M300 S2959 P150
M300 S2793 P150
M300 S2489 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S1661 P150
M300 S880 P150
M300 S2093 P150
M300 S0 P150
M300 S880 P150
M300 S2093 P150
M300 S2349 P150
M300 S0 P300
M300 S2489 P300
M300 S0 P150
M300 S2349 P150
M300 S0 P300
M300 S2093 P300;
M73 P100 R0

leptun, 2019-12-11:
@WhiteDragonD A fix for this issue is already developed for 3.9 and it’s on its way to get merged in the firmware. Thanks for reporting.

leptun, 2019-12-11:
Pr with the fix: https://github.com/prusa3d/Prusa-Firmware/pull/2149

WhiteDragonD, 2019-12-11:
Awesome news, thanks!

PR #2149 (2020-02-04): PFW-1056: Re-enable Tone on timer4
This PR reenables tone functionality. This is not enough by itself though, that's why there is another PR here for the other required changes. https://github.com/prusa3d/Arduino_Boards/pull/3
Closes #2365
Closes #2229
Closes #1995
Closes #2432
closed 2020-02-04 by DRracer