prusa3d/Prusa-Firmware #2373: Endstop gets ignored by prusa while moving

kyjak, 2019-12-16:
MK3 3.8.1

I want the Prusa to stop movement when hitting an endstop.
I connected a limit switch to the X axis and configured it properly.
I commented out // #define ENDSTOPS_ONLY_FOR_HOMING // If defined the endstops will only be used for homing
I checked if the endstop works issueing M119 (it behaves in a correct way)

yet the MK3 ignores the limit switch and goes on with movement even when triggering it.
Is this normal or do we hve a bug?
  
**To Reproduce**
Just follow what I did.

**Expected behavior**
Obey endstops while processing G0/G1 moves
  
**G-code**
M119 ; to double check the probe settings
G0 X10 Y10 F100

vintagepc, 2019-12-17:
You probably also need to disable sensorless homing; I don't think that and physical endstops mix well.

kyjak, 2019-12-18:
I had to fork the firmware and make more tweaks including a special gcode command to switch between sensorless and driver endstop behavior.
Are you interested in such code so I make a PR? 
This basically turns the printer into a 3D scanner using a probe attached to X limit switch.

closed 2023-10-12 by github-actions[bot]