prusa3d/Prusa-Firmware #2374: [BUG] Connecting Repetier Server to USB port kills running print from SD card

vmario89, 2019-12-16:
**Printer type** MK3S
**Printer firmware version** 3.8.1

Hi,
it figured out that my MK3S printer stops the current print from SD card without any recoverable state if i connect a raspberry pi to USB port of the printer. On my raspi a Repetier Server is running to send jobs to Prusa MK3s (and other printers).

Problems in my practical usecase: 
- if i print something and in the middle of the print somebody else connects the printer, the print will fail. Thats not the desired result :-(
- I also can have the printer already attached to USB before printing from SD card and the if i print from SD this works fine. But maybe if the connected Raspi reboots due to some error or power loss, the print will fail again because the printer re-inits at the new found USB connection
- its not an option to switch to Octoprint for me (i am running up to 7 other printers on the same hub)

Question: does anybody know if this happens with Octoprint too?

Is it something to fix in firmware or in the 3d printer server software? or both?

regards, Mario

leptun, 2019-12-16:
This is the desired behavior of the hardware. Whenever the usb serial port is opened, the reset line is toggled as to allow for firmware update. This can only be disabled in hardware (trace cut on the pcb) or through some software hacks on the PC. 
There might also be a special command to send on the serial for the usb-to-serial chip to disable this functionality, but updating the firmware will be really hard after that.

RolfHodl, 2019-12-18:
Octoprint is the same behavior. First time experience was bad - just clicked on the Octopod App by accidient on connect and the print was gone :/ -> printerrestart - no physical conection - just a software connect.

mkbel, 2019-12-22:
This is feature. There is workaround: https://github.com/prusa3d/Prusa-Firmware/issues/1572#issuecomment-481477649

closed 2019-12-22 by mkbel

jacopotediosi, 2024-03-01:
I just released a new plugin for OctoPrint that might help solve this problem. [Prusa Reset Mode](https://github.com/jacopotediosi/Octoprint-PrusaResetMode) adds a tab to Octoprint, with buttons that allow you to send serial commands directly from the OctoPrint interface to enable and disable this "automatic reset feature".