prusa3d/Prusa-Firmware #2375: After firmware update, Fw. 3.8.1 printer has a Z-axis error

Hayasiggi, 2019-12-17:
Hello everybody!

After I updated my printer, I have an original Prusa i3 mk3s mmu2s, from Fw. 3.7.1 to 3.8.1 the following error occurred.

The printer makes an unnecessary downward movement of the Z axis several times during printing and thereby destroys the printed object

I did complete resets with Fw. 3.8.1, 3.8.0, 3.7.2 all without success. The negative Z error still exists. Now I print again with Fw. 3.7.1 and have no problems.

I used the same test object for all prints that was created with PrusaSlicer 2.1.0 Win 64.

I had this error before, with Fw. Vers. 3.6.0 on March 19, 2019.
with Fw. Vers. 3.7.0 RC 1 and with ver. 3.7.1 the error does not occur.

I have also updated the MMU2, this update version 1.0.6 does not cause any problems since I do it too
with Fw, 3.7.1 use.

I have an avi. Video from 19.03.2019 attached, at about position 13:15:47 the error occurs.

[3d Drucker Z-Achsen Crash.zip](https://github.com/prusa3d/Prusa-Firmware/files/3974938/3d.Drucker.Z-Achsen.Crash.zip)

[G-Code Sektglas.zip](https://github.com/prusa3d/Prusa-Firmware/files/3974960/G-Code.Sektglas.zip)

leptun, 2019-12-18:
@wavexx The entire planner seems to be corrupted multiple times during the print, the Z axis movement only being in the last of these failures. In the other cases, only Y seemed to move towards MIN for a few times. Any idea what might trigger such an issue?
@Hayasiggi Are you printing from octoprint or from the SD card when the error occurs? It might be some corruption of the gcode from octoprint to the printer that triggers the issue.

wavexx, 2019-12-18:
From the video, it looks like the printer also makes several unnecessary Y moves. It's hard to tell from the camera; does this happen when the stem finishes and the glass cup gets larger?

Knowing whether this is printed from the SD and/or from octoprint would be very useful.

Hayasiggi, 2019-12-19:
I don't print with Octoprint.
The printer shows this behavior even when printing from sd.
I print with Repetier-Host server.
You saw correctly the printer also makes a Y and X.
Axis movement as if it has to orientate itself, but that doesn't really matter unless the print takes a little longer.

The downward movement happens involuntarily and cannot be predicted.
Sometimes 2-3 times in a row.
Sometimes just before the end of printing.
But never on the first two shifts.

mdaneman, 2019-12-21:
I have exactly the same problem with 3.8.1 (as I reported in issue #2358 ).  I haven't seen the z-crash, but the extruder makes unncessarily moves in X and Y.  Reverting to 3.8.0 makes the issue go away.

Hayasiggi, 2019-12-21:
@mdaneman,Thank you, I will then flash again to 3.8.0 and continue testing.

Hayasiggi, 2019-12-21:
@mdaneman Unfortunately it didn't work. The errors are still there

wavexx, 2019-12-21:
@Hayasiggi do you have any smaller gcode where this happens?

Hayasiggi, 2019-12-21:
@wavexx Not at the moment, I will create a small object
and test whether the error also occurs here.
if so i will put the g-code here

wavexx, 2019-12-21:
@mdaneman Issue #2358 looks very similar, and this looks like happens already on the first layer. Does it happer every time, or seems to be more random? A video would be very helpful.

Hayasiggi, 2019-12-21:
@mdaneman The additional X, Y movements take place from the first level, but not always.
The z downward movement usually occurs later from 15 levels.
but then randomly.
very often before the end of printing
I will try to make a video on which you can see everything.

Making videos is not my thing, so it can take a while.

Hayasiggi, 2019-12-21:
@mdaneman At the beginning I had attached a video.
you should be able to see everything there.

wavexx, 2019-12-21:
> @wavexx At the beginning I had attached a video.
> you should be able to see everything there.

Yes, I've seen this and that's very helpful. I was hoping I could also see the behavior @mdaneman is having.

mdaneman, 2019-12-22:
> > @wavexx At the beginning I had attached a video.
> > you should be able to see everything there.
> 
> Yes, I've seen this and that's very helpful. I was hoping I could also see the behavior @mdaneman is having.

Unfortunately I didn’t capture a video of the behavior before I switched back to 3.8.0 which, for me, makes the problem go away.  It basically does the following:  the printer is printing, then the extruded travels to the edge of the build plate in either x or y, then It comes back to more or less where it was and continues printing, leaving a blank line since it didn’t print while it was doing its extra move. You can see these blank spots as well as bits of filament it left on the build plate during the random move in the picture I attached to my issue report. I’m not sure if it only happens on the first layer. I think I’ve seen it on higher layers as well, but not 100% sure. Also the random moves timing and location are not consistent between prints.

wavexx, 2019-12-23:
@mdaneman I tried to print two times your gcode (FiveSquare) using FW 3.8.1, but couldn't reproduce the issue.

Does it happen at every print for you?
Does it happen if you print right after a reset?

Looking again at your picture in #2358, it looks like the gap starts exactly at the infill boundary. I debugged a very similar issue during the development of Linear Advance 1.5 and I wonder if this could be related.

Also for @Hayasiggi, in the Filament Settings > Custom G-Code, try removing everything and explicitly disable linear advance instead by writing ``M900 K0`` and reprint.

Although I do not see anything specific between 3.8.0 and 3.8.1 that would cause this to change, the current linear advance code is very time sensitive and there are a number of things that could make it happen more frequently. Also, testing this would help in excluding at least one class of issues.

closed 2023-10-04 by github-actions[bot]