prusa3d/Prusa-Firmware #2377: [FEATURE REQUEST] Maintenance intervals

Sc0tty83, 2019-12-18:
It would be a nice feature to get a maintenance warning after x hours of printing.

For example to retighten all the screws, checking belt tension, cleaning and oiling all shafts and so on. Just like in cars that demand an oil change or inspection every x miles/kilometres.

3d-gussner, 2019-12-20:
Like the idea, every environment might be different, dusty place and so on. But in general it is something that could be useful.
- Print time xxxx triggers a message
- Message shouldn't block the printer but at least delay it for a moment every time it starts up
  - until it is acknowledged that it was done
For the users don't want this
- a menu may be useful to enable this feature or not

3d-gussner, 2019-12-20:
@Sc0tty83 The Feature issue template is there for a reason, so please try next time to fill it in.

closed 2023-10-12 by github-actions[bot]