prusa3d/Prusa-Firmware #2378: [Feature Request] Randomize Part Location on Bed

PotatoFi, 2019-12-20:
**Current Scenario**

Repeat parts always print in the same spot on the print bed, causing wear on certain parts of the bed.

**Done When**

An option is available to randomize the location of the part on the bed, which will help evenly spread the wear out all over the bed.

leptun, 2019-12-20:
This is not really possible to do in firmware since the firmware doesn’t know how big the printed object is and can’t determine a safe offset that won’t go over the edge of the bed. All gcode instructions for the firmware are absolute coordinates and it would mean we always have to compensate for the offset, something which will add overhead to the processing speed of the gcode and may cause hard to debug consequences for crashDetect and especially powerPanic.
Also, I don’t see why normal users would want such a feature. The wear on the steel sheets is not significant enough to necesitate moving the part around the bed. The only scenario wherw you might want something like this is if you print a lot of PETG on the smooth sheet. All other scenarios either don’t wear the bed or are problematic if not close to the center of the bed (ABS/ASA, NYLON, PC) and may show uneven warping.

leptun, 2019-12-20:
@3d-gussner Hmmm. The github bot didn’t add the “Feature request” tag to this issue.

ochm, 2020-01-19:
IMHO the firmware does what the slicer prepare, no further intelligence is required.

closed 2023-10-12 by github-actions[bot]