prusa3d/Prusa-Firmware #2379: Mk2.5S filament use statistic reset after fw update and not counting

hagan66, 2019-12-21:
I have updated my mk2s to mk2.5s with fw 3.8.1 and now the filament use statistic is reset to zero and does not record any material use. I have reflashed the fw, no luck.
  
Not a big problem, but I thought you would like to know. :-)

leptun, 2019-12-21:
@hagan66 Please try clearing th statistics in the factory reset menu.

Also, are you printing from octoprint (or anything other than the SD)?

hagan66, 2019-12-21:
I might have used octoprint only since the update, not sure. Is that the problem? :-)

leptun, 2019-12-21:
@hagan66 at the moment the statistics only count if you print from the SD card. I might fix the issue at a later time.
I’m pretty sure this issue is a duplicate, but I’ll have to search for the other one

hagan66, 2019-12-21:
Sorry for posting duplicate. Im on my phone and could not find search. So i browsed a lot of pages but did not find it. Thanks for tje superfast reply!

closed 2019-12-21 by hagan66

leptun, 2019-12-21:
@hagan66 No problem. I’m also on my phone. I’ll search later when I get to my laptop and link it here.